tbnobody/OpenDTU #1114: [Request] Overflow /0/yieldday at 64000 Wh for HM600 permanently supplied by battery

daebikelinz, 2023-07-04:
### Is your feature request related to a problem? Please describe.

When the inverter is permanently supplied by a battery it doesn't get unpowered to store the yield_day.
I think that an overflow happens at a yield_day of 64 000 Wh for each of the two strings. That should be the reason why it reseted two times slightly delayed.
For now i was only tracking the yield_day of channel 0 (sum of the two strings) and the problem occured a little under 128 000 Wh.
Because of the slightly unequal physical connection, there is some difference for the two strings in terms of slighlty different power delivery.
It also messes up the /0/yieldday (Energy total in the picture) variable.

[image]

### Describe the solution you'd like

A setting that can be enabled in openDTU to force a simulated "power down" to store the yield_day values and reset it to 0 Wh.
I'm honestly not sure if my problem is solvable because the software of the inverter was never developed for a permanent supply.

### Additional context

By the way thank you for the brilliant software.
My HM600 grid PV system with a battery is phenominal!
My remaining grid power is regulated to zero nearly the whole day besides the power peaks over 600 Watts.

[image]

DPO99, 2023-07-04:
i think this is not possible because the HM is sending this parameter. Why do you dont restart the inverter at every night?

daebikelinz, 2023-07-05:
Ok sure i should have taken another look at the documentation. Thank you for that hint!
"[serial]/cmd/restart | W | Restarts the inverters (also resets YieldDay) | 1"

[image]

That worked perfect and the inverter was only offline for a few seconds.

closed 2023-07-05 by daebikelinz