tbnobody/OpenDTU #1117: HM1500 plötzlich im Powerlimit 200W

mrstackit, 2023-07-07:
### What happened?

Hallo, sporadisch wird vom System ein Powerlimit 200W gesetzt. Ich habe schon geprüft ob dies per MQTT gesetzt wird indem ich die cmd topics abgeschaltet habe. Das verhalten bleibt trotzdem. Auch wenn ich einmal das Powerlimit dauerhaft auf 100% stelle passiert es wieder sporadisch das es auf 200W gestellt wird. Bis jetzt ist dies morgens direkt oder zur Mittagszeit passiert.

Kann ich dagegen was tun? Kann ich irgendwelche Logs liefern damit das geprüft werden kann?

### To Reproduce Bug

-

### Expected Behavior

-

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.6.28

tbnobody, 2023-07-07:
> Kann ich irgendwelche Logs liefern damit das geprüft werden kann?

Ja, man bräuchte vom Zeitpunkt an dem das Limit gesetzt wird entweder das [Serielle Log](https://github.com/tbnobody/OpenDTU/wiki/FAQ#q-how-to-observe-the-esp32-boot-process-by-looking-at-the-serial-console) oder das Log der Web Konsole (Info --> Konsole).

mrstackit, 2023-07-08:
Ok ich werde ein Serielles Log erstellen. Ich muss dafür aber erst die OpenDTU anders anschließen damit ich über längeren Zeitraum ein Log erstellen kann. Das Problem ist heute auch wieder aufgetreten andere Uhrzeit aber, daher denke ich das es dann auch im Log zusehen sein wird.

mrstackit, 2023-07-11:
Im Seriellen Log konnte ich jetzt folgends finden was ich aber merkwürdig finde da ich nur MQTT lese und die Instanz für das Werte senden abgeschaltet habe.

Received MQTT message on topic: solar/SN/cmd/limit_nonpersistent_relative
Limit Non-Persistent: 100 %
Received MQTT message on topic: solar/SN/cmd/limit_nonpersistent_absolute
Limit Non-Persistent: 200 W

Ich habe jetzt die MQTT Instanz gelöscht und die cmd topics auch. Ich werde beobachten ob es weiterhin vorkommt und mich dann nochmal dazu melden

mrstackit, 2023-07-22:
Seitdem ich die MQTT Instanz für das Werte schreiben im iobroker gelöscht habe ist das Problem nicht mehr vorhanden. 

Als Tipp für andere die das mitlesen und ein ähnliches Problem haben kann ich nur mitgeben die MQTT Instanz bei Problemen nicht nur zu Stoppen sondern auch zu löschen. Die Werte wurden einfach willkürlich gesetzt ohne das ich es manuell habe oder ein script es angestoßen hat.

closed 2023-07-22 by mrstackit