tbnobody/OpenDTU #1118: [Request] Usage of relative paths

basi0, 2023-07-07:
### Is your feature request related to a problem? Please describe.

I'm trying to implement OpenDTU behind my existing reverse proxy. 
Therefore i set up a virtual directory, https://internal.mydomain.com/dtu, which is reverse proxied to the OpenDTU root.
Unfortunately OpenDTU expects to be accessed in the root directory, so it doesn't work. E.g. the application javascript is absolutely adressed: <script type="module" crossorigin src="**/js/app.js**"></script>

Maybe it's possible to change OpenDTU to use relative paths instead, or to set an alternative web root in the configuration?

### Describe the solution you'd like

Change OpenDTU to use relative paths instead, or to set an alternative web root in the configuration

tbnobody, 2023-07-07:
Even if I would change the path in the `index.html` to a relative path it would still fail to load because all the API calls access the root path (e.g. `/api/liveview/status`). And if this could somehow be fixed (to be honest, I correctly don't have any idea how) the next problem is the websocket which delivers the live view and console data.

jstammi, 2023-07-09:
I will have a look to it

ahinrichs, 2023-07-16:
Your proxy should be able to handle this. With nginx use a slash at the proxy_pass URI (location /dtu/ { proxy_pass http://opendtu/; })

allcoolusernamesaregone, 2023-07-28:
Websockets shouldnt be a problem.   i'm using EVCC behind nginx as an reverse proxy. EVCC also uses websockets. You can force an upgrade to an http conenction, e.g.

...
      proxy_pass http://192.168.22.111:7070/;
      proxy_redirect off;
...
      proxy_http_version 1.1;
      proxy_set_header Upgrade $http_upgrade;
      proxy_set_header Connection "upgrade";
...

Last two rows do the trick...

allcoolusernamesaregone, 2023-07-28:
... and yes, this change to relative paths within the UI would be really great.

jstammi, 2023-07-29:
I had a look. It is no prob wrt websocket and resources being loaded, this is easy.

But the framework used for the ui (vue.js) causes troubles. It generates the html doc clientside. And had to consider any path that somehow has to get known to it. It COULD bei possible, I saw some options. But I failed to test it up to now, mostly due to lack of time actually.

stefan123t, 2023-07-29:
Having had a quick peak on this Internet thing, should be possible:

[How to change basepath on deployment of Vue app](https://stackoverflow.com/questions/53162491/how-to-change-basepath-on-deployment-of-vue-app)

On file vue.config.js
```typescript
module.exports = {
  /* ... */
  publicPath: process.env.NODE_ENV === 'production' ? '/my-app/' : '/'
}
```

On file router.js
```typescript
/* ... */
import { publicPath } from '../vue.config'
/* ... */
export default new Router({
    mode: 'history',
    base: publicPath,
    /* ... */
})
```

jstammi, 2023-07-29:
Hm, not really like this. As titled "How to change basepath ON DEPLOYMENT of Vue app". That is when @tbnobody builds the release.
It is needed to be able to specify this at RUNTIME. AFAIU there should be some possibility. E.g. like described here:
https://medium.com/@gentritabazi01/vue-js-make-publicpath-dynamic-33c065960eb9

But up to now I lack the time to setup a test environment to verify this working.

allcoolusernamesaregone, 2023-07-30:
I would be happy to test this. All I need is either a bin file to flash or either some patched files of a specific branch or tag, ot the name of the branch/tag to use to build an image..

stefan123t, 2024-07-10:
@jstammi thanks for confirming that it ain't no issue for the websocket and rpc API calls to modify the WebServer listeners.

Regarding the suggested change of the `__webpack_public_path__` the link you mentioned IMHO uses some **pre-processor options/magic** which render this global variable `:writable` and export the symbol for us to be able to overwrite it in the DOM model of the browser.

Here you will find quick steps to make `publicPath` dynamic.

#### Step 1: Go to `public/index.html` and define a global variable.

```javascript
<script type=”text/javascript”>window.PUBLIC_PATH = ‘/’;</script>
```

#### Step 2: Create `public-path.js` file inside `src` folder and write code like below:

```javascript
/* global __webpack_public_path__:writable */
/* exported __webpack_public_path__ */
__webpack_public_path__ = window.PUBLIC_PATH;
```

#### Step 3: Open `main.js` file and `import public-path.js` file at the top of `main.js`:

```javascript
import ‘./public-path.js’
```

p1ngb4ck, 2025-07-25:
Stumbled upon this, when I was trying to add OpenDTU-WebIf as a HA dashboard for admins.
Yes, configurable relative Web-Root-Path would be indeed a needed and wanted feature. Could that be added as a wanted feature please ?