tbnobody/OpenDTU #1120: Funkchip nach update der aktuellen firmware nicht mehr erkennbar

arnjan, 2023-07-07:
### What happened?

nach dem update kann der installierte Funkchip nicht mehr erkannt werden und die Daten nicht ermitttelt werden

### To Reproduce Bug

after firmware upgrade OTA

### Expected Behavior

lesbare Daten und Funkchip

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

v23.6.21 

### Relevant log/trace output

```Shell
OpenDTU-B8E6A0
SDK-Version	v4.4.4
Konfigurationsversion	0.1.25
Firmwareversion / git Hash	v23.6.21
PIO Umgebung	generic_esp32
```

### Anything else?

no

arnjan, 2023-07-07:
It was a problem of vscode, somehow it changed from generic to generic_esp32 al platformiio env. After manually changing back to generic it worked again

tbnobody, 2023-07-07:
Yes, that is the intention of the generic_esp32 profile... it does not have any pre-defined pin assignments. you can flash it on every esp32 based board and just upload an device profile matching your board or needs. I will close this issue as it is not really an issue but it works as intended... Please see also https://github.com/tbnobody/OpenDTU/wiki/Binary-Installation

closed 2023-07-07 by tbnobody