tbnobody/OpenDTU #1123: Post live Data to Web Server

MaximilianSteverding, 2023-07-08:
### Is your feature request related to a problem? Please describe.

I am not working with Home Assistant but would like to collect and process the live data of my Inverter on the web.

### Describe the solution you'd like

It would be great if we could implement a section in the OpenDTU where one could write down a web-address to which the OpenDTU posts a JSON File or a simple Get-Request with the current live data, timestamp and location of the DTU. It could be triggered when the DTU received new values from the inverter.

### Describe alternatives you've considered

Of course one could also first get the values through MQTT and oder load it to a server but that would only add the necessity to get another device in the line. I think it would be easier to include that feature directly in the OpenDTU.

### Additional context

Looking forward to your opinions and help!

Kampfschildkroete, 2023-07-15:
I think for such cases you can use the webAPI (described [here](https://github.com/tbnobody/OpenDTU/blob/master/docs/Web-API.md)). Then you can fetch the data on your on own with your own defined interval?

JonasGhost, 2023-09-01:
I have some use cases where I need a cloud. It would be possible to use MQTT over the WAN, I would prefer UDP some how useful encrypted or HTTP.

There are already cloud solutions that can handle HTTP POST and GET request. e.g. https://pvoutput.org/

Accessing the HTTP interface of openDTU over a WAN connection without VPN is not the solution for my use cases.

stefan123t, 2023-10-14:
I would also like to log data to my nextcloud (WebDAV) share from the openDTU. Question would be if we can first check the document already exists using a HEAD request and in case it exists append using the UPDATE request. If it is new we should initialize the file using PUT request.

stefan123t, 2023-10-14:
The SabreDAV server (this is used e.g. by Nextcloud / Owncloud Open Source servers) does allow adding single lines to an existing file using the PATCH request method. This method described here: https://sabre.io/dav/http-patch/

The following HTTP(S) request will append the line 1234 to the /file.txt on the server.
```http
PATCH /file.txt
Content-Length: 4
Content-Type: application/x-sabredav-partialupdate
X-Update-Range: append

1234
```

simsasaile, 2023-10-18:
I would also appreciate this feature, I would like to send the data directly to an emonCMS webserver (from OpenEnergyMonitor). Maybe Tasmota scripts could serve as inspiration for this: https://tasmota.github.io/docs/Scripting-Language/
This is super flexible to configure for all webservers (PVoutput would be easily addressable with this as well).

stefan123t, 2024-07-10:
Similar to the above `PATCH` method in **SabreDAV / Nextcloud** which made little progress until now,
here is a guide to add rows with metric results to a **Google Sheet** in a Project.

FWIW this is probably a bit more than the average CSV file, but it may be what others want to work with ?

[ESP32 Datalogging to Google Sheets (using Google Service Account)](https://randomnerdtutorials.com/esp32-datalogging-google-sheets/)

```c++
//Google Project ID
#define PROJECT_ID "REPLACE_WITH_YOUR_PROJECT_ID"
// Service Account's client email
#define CLIENT_EMAIL "REPLACE_WITH_YOUR_CLIENT_EMAIL"
// Service Account's private key
const char PRIVATE_KEY[] PROGMEM = 
"-----BEGIN PRIVATE KEY-----\n
REPLACE_WITH_YOUR_PRIVATE_KEY\n
-----END PRIVATE KEY-----\n";
// Google Spreadsheet ID
const char spreadsheetId[] = "YOUR_SPREADSHEET_ID";

setup() {
  // Set the callback for Google API access token generation status (for debug only)
  GSheet.setTokenCallback(tokenStatusCallback);
  // Set the seconds to refresh the auth token before expire (60 to 3540, default is 300 seconds)
  GSheet.setPrerefreshSeconds(10 * 60);
  // Begin the access token generation for Google API authentication
  GSheet.begin(CLIENT_EMAIL, PROJECT_ID, PRIVATE_KEY);
...
}

loop() {
  FirebaseJson valueRange;
  valueRange.add("majorDimension", "COLUMNS");
  valueRange.set("values/[0]/[0]", epochTime);
  valueRange.set("values/[1]/[0]", temp);
  valueRange.set("values/[2]/[0]", hum);
  valueRange.set("values/[3]/[0]", pres);
  
  // Append values to the spreadsheet
  bool success = GSheet.values.append(
     &response     /* returned response */, 
     spreadsheetId /* spreadsheet Id to append */, 
     "Sheet1!A1"   /* range to append */, 
     &valueRange   /* data range to append */);
  if (success) {
      response.toString(Serial, true);
      valueRange.clear();
  }
  else { Serial.println(GSheet.errorReason());
  }
...
}
```