tbnobody/OpenDTU #1125: Persistent und Non persistent Limits können nicht unabhängig gesetzt werden

johi-1, 2023-07-08:
### What happened?

 bei mir wird das Limit Persistent und  Limit Non persistent immer gleichzeitig verändert.  Ich verwende Open DTU und Home Assistant. Aufgefallen ist es mir  als ich mittels  Node Red eine 'Null Einspeisung' realisieren wollte. Obwohl ich nur einen Befehl an den Wechselrichter schickte  (Limit Non Persistent) habe ich bei den Entitäten in HA gesehen, dass auch das Persistent Limit verändert wurde.  Ich habe dann in HA die Veränderung getestet und bemerkt , dass sich beide Limits ändern wenn ich eines ändere.   Ist das ein Software Bug oder  kann etwas bei mir im System falsch sein?

### To Reproduce Bug

[image]

### Expected Behavior

Es sollte sich nur einer der beiden Werte änder

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.6.28

tbnobody, 2023-07-08:
Dies ist kein Bug oder Fehler deinerseits. Es ist schlicht und ergreifend eine Einschränkung des Inverters. Man kann ein Limit zwar als Persistent oder Non-Persistent setzen aber anschließend nicht mehr auslesen ob es nun Persistent oder Non-Persistent ist. Man kann (nach aktuellem Wissensstand) einfach nur ein Limit lesen. Daher sind beide Werte auch identisch.
OpenDTU speichert selbst keine Einstellungen des Wechselrichters sondern liest diese jedes Mal nur aus.

closed 2023-07-08 by tbnobody

johi-1, 2023-07-08:
Danke, für die schnelle Antwort.

johi-1, 2023-07-09:
Hallo, habe noch eine Frage.

Wenn ich ein Non Persistent Limit setze, danach den Inverter ausschalte und dann wieder einschalte müsste er doch eigentlich wieder  das  original persistent Limit zeigen.

Macht er aber nicht.  (Persistent und Nicht Persistent bleiben auf z.B. 80%, obwohl ich nur Nicht Persistent geändert hatte)

Mein Sohn verwendet Ahoy. Bei ihm verändert sich in HA nur das Limit , das er ändert. 

Ich verstehe das nicht.

Viele Grüße

JH 

Von: tbnobody
Gesendet: Samstag, 8. Juli 2023 17:59
An: tbnobody/OpenDTU
Cc: johi-1; Author
Betreff: Re: [tbnobody/OpenDTU] Persistent und Non persistent Limits können nicht unabhängig gesetzt werden (Issue #1125)

Dies ist kein Bug oder Fehler deinerseits. Es ist schlicht und ergreifend eine Einschränkung des Inverters. Man kann ein Limit zwar als Persistent oder Non-Persistent setzen aber anschließend nicht mehr auslesen ob es nun Persistent oder Non-Persistent ist. Man kann (nach aktuellem Wissensstand) einfach nur ein Limit lesen. Daher sind beide Werte auch identisch.
OpenDTU speichert selbst keine Einstellungen des Wechselrichters sondern liest diese jedes Mal nur aus.

tbnobody, 2023-07-09:
> Wenn ich ein Non Persistent Limit setze, danach den Inverter ausschalte und dann wieder einschalte müsste er doch eigentlich wieder  das  original persistent Limit zeigen.

Das kommt darauf an... Folgendes Beispiel:
* Du setzt das Limit Persistent auf 50%
* Dann Non-Persistent auf 80%

Nach einem Aus-/Einschalten wird das Limit wieder auf 50% sein, denn das war das letzte Persistente Limit.

johi-1, 2023-07-09:
Ich habe eben das Limit Persistent auf 100% gesetzt dann das Limit non Persistent auf 50%.
Nach dem Ausschalten , und wieder Einschalten des Wechselrichters war die Anzeige beider Limits immer noch bei 50%

Ich konnte aber dann nach ‚längerer‘ Zeit (ca. 2min) sehen, dass beide Limits wieder auf 100% sprangen.

Ich hatte vorher wohl nicht lange genug gewartet.

Danke noch einmal.

Alles ok.

Wirklich eine tolle Software die ihr da fabriziert habt.

Von: tbnobody
Gesendet: Sonntag, 9. Juli 2023 11:40
An: tbnobody/OpenDTU
Cc: johi-1; Author
Betreff: Re: [tbnobody/OpenDTU] Persistent und Non persistent Limits können nicht unabhängig gesetzt werden (Issue #1125)

Wenn ich ein Non Persistent Limit setze, danach den Inverter ausschalte und dann wieder einschalte müsste er doch eigentlich wieder das original persistent Limit zeigen.
Das kommt darauf an... Folgendes Beispiel:
• Du setzt das Limit Persistent auf 50%
• Dann Non-Persistent auf 80%
Nach einem Aus-/Einschalten wird das Limit wieder auf 50% sein, denn das war das letzte Persistente Limit.

Omega13x, 2023-07-09:
Zwischen Ausschalten und Ausschalten gibt es einen Unterschied. Wenn du den WR stromlos machst, in der Regel nachts, wird beim Wiederanfahren das Limit persistent genommen, wenn nicht gleich eine Regelung greift. Einfach über die Schalterfunktion der Gui ein- und auszuschalten bringt nichts, weil der letzte gesetzte Wert immer noch im Speicher ist. Nur stromlos setzt automatisch persistent.

johi-1, 2023-07-09:
Danke, hatte ich mir gedacht und zusätzlich Stecker gezogen.

Von: Omega13x
Gesendet: Sonntag, 9. Juli 2023 13:58
An: tbnobody/OpenDTU
Cc: johi-1; Author
Betreff: Re: [tbnobody/OpenDTU] Persistent und Non persistent Limits können nicht unabhängig gesetzt werden (Issue #1125)

Zwischen Ausschalten und Ausschalten gibt es einen Unterschied. Wenn du den WR stromlos machst, in der Regel nachts, wird beim Wiederanfahren das Limit persistent genommen, wenn nicht gleich eine Regelung greift. Einfach über die Schalterfunktion der Gui ein- und auszuschalten bringt nichts, weil der letzte gesetzte Wert immer noch im Speicher ist. Nur stromlos setzt automatisch persistent.