tbnobody/OpenDTU #1128: [Request] Turn LED 1 (inverter indication) off when in night-mode

IdrinkOban, 2023-07-08:
### Is your feature request related to a problem? Please describe.

When OpenDTU is in night mode, the LED 1 (inverter indication) should be turned off.
Currently, the LED starts blinking, when inverters stop producing and turns off, when the first inverter becomes unreachable. 
So far, so great! :-)

But a soon, as the inverter/NTP is in night mode, the LED turns on again.
Seems seems to me not very logical.

### Describe the solution you'd like

When OpenDTU is in night-mode, LED 1 (inverter indication) should be turned off.

### Describe alternatives you've considered

An alternative is, to enable "Poll inverter data at night" for at least one inverter.
This will generate the behavior described by me.

### Additional context

Thanks for OpenDTU. This product is absolutely great!

tbnobody, 2023-07-09:
>  When OpenDTU is in night-mode, LED 1 (inverter indication) should be turned off.

It's not so easy.... Imagine you have multiple inverters. One  shall be polled at night, whereby the other should not... 
Therefore it would not make any sense to turn off the inverter at night. 

Currently the LED shows whether all inverters are in it's correct state. The LED is ON if all enabled inverters are reachable and are producing. If no inverter is enabled this state is interpreted as  true... because on the other hand, there is no non-reachable inverter in the inverter in the list of enabled inverters (which is a empty list at this momemt to be honest).

Maybe it does also make sense to turn of the LED if no inverter is currently enabled.

IdrinkOban, 2023-07-09:
> > When OpenDTU is in night-mode, LED 1 (inverter indication) should be turned off.
> 
> It's not so easy.... Imagine you have multiple inverters. One shall be polled at night, whereby the other should not... Therefore it would not make any sense to turn off the inverter at night.

Ah, I haven't considered this. I do have 4 converters, but I have configured them all in the same way (don't poll at night).

> If no inverter is enabled this state is interpreted as true... because on the other hand, there is no non-reachable inverter in the inverter in the list of enabled inverters (which is a empty list at this moment to be honest).
> Maybe it does also make sense to turn of the LED if no inverter is currently enabled.

That could indeed make sense. At least from my point of view.

DejanBukovec, 2023-07-10:
I vote for change that LED1 is turned OFF when mode change to "Night" and there is no inverter with enabled "Poll inverter data at night".

tbnobody, 2023-07-10:
Implemented in fe3d6588bc5d114d8c42097116033da917fb0a80

closed 2023-07-10 by tbnobody