tbnobody/OpenDTU #1147: [Request] - ESP restart via MQTT

Phantomias2006, 2023-07-15:
### Is your feature request related to a problem? Please describe.

Hello,

I have integrated openDTU into NodeRed using MQTT.
In case of an error I would like to restart the ESP.
This function is not available via MQTT. 

### Describe the solution you'd like

Add a new MQTT topic for an ESP restart

DPO99, 2023-07-21:
Which error do you receive, that you need a restart?

Phantomias2006, 2023-07-21:
Hello, thank you for your reply.

I use the project „openDTU on battery“ and there the CAN data of the Pylontech battery storage and the Huawei power supply are read. In case of longer connection problems I would do a restart as fallback. 
Because the restart is included in the WebAPI and missing in MQTT I created a feature request in the main project.

kjmwwi, 2023-08-07:
This reboot feature would be also usefull, if there is no response of opendtu because of "flooded" mqtt-queue because of to much power limit commands

crisi-solar, 2024-06-20:
Da ich jeden Tag die openDTU neu starten muss wäre es toll wenn ich es über MQTT in Homeassistant machen kann.
Warum muss ich jeden Tag neu starten? Siehe:
https://github.com/reserve85/HoymilesZeroExport/issues/211

Omega13x, 2024-06-20:
Warum musst du die DTU jeden Tag neu starten?

crisi-solar, 2024-06-20:
@Omega13x Siehe Link oben

stefan123t, 2024-08-05:
@Phantomias2006 @kjmwwi @crisi-solar a topic to restart the OpenDTU will have to defined as an additional MQTT topic in #759.

We would need to agree on a common OpenDTU / Ahoy MQTT topic tree / structure with a writable / control topic for DTU restart / reset here first.

roobbb, 2026-01-07:
Hello.

I wish you all a happy new year.

Thank you for your quick response. I tried posting to the endpoint as mentioned above. Even with authentication. But unfortunately it did not work. 
I think I will "solve" it by an extra wifi power socket before my OpenDTU's power supply and switch that one on and off. It's a bit overkill because this could just be done in software, but as long as it is an unwanted feature such "workaround" does the job too.

Thank you and kind regards
roobbb

----
What I tried:
```
curl http://<myip>/api/maintenance/reboot
curl -u admin:<mypassw> http://<myip>/api/maintenance/reboot --verbose
curl -u "admin:<mypassw>" http://<myip>/api/maintenance/reboot -d 'data={1}' --verbose
curl -X POST -u "admin:<mypassw>" http://<myip>/api/maintenance/reboot --verbose
curl -u "admin:<mypassw>" http://<myip>/api/maintenance/reboot -d 'data={"serial":"<myserialno>","reboot":"1"}' --verbose
```

mathieucarbou, 2026-09-19:
> I use the project „openDTU on battery“ and there the CAN data of the Pylontech battery storage and the Huawei power supply are read. In case of longer connection problems I would do a restart as [...]

> This reboot feature would be also usefull, if there is no response of opendtu because of "flooded" mqtt-queue because of to much power limit commands

These are not good reasons.
I think you should instead open an issue in the underlying projects.
A MCU should not require a restart because of a normal use.
Here you are describing bugs (resource depletions) that should be fixed.

"Too much power limit commands"

=> the regulation algorithm is too fast and does not account for the slow communication (timeout)

You might want to try the fix I did in #3225 to fasten the inverter com: I can send and apply in less than 2 seconds some limits on my inverters (8 HMS). Before, it was taking about 16 seconds to complete.

ss89, 2026-10-25:
Sieht so aus als wäre das jetzt integriert seit dem september release (d3f7d0716cce87b2af0542e6453113c5e6a51821). Danke an @stefan123t und @tbnobody!

Ich bin gespannt es nächstes mal auszuprobieren, sobald sich mein (Funk-)Modul aufhängt.

Erwin40, 2026-10-25:
Yes, funktioniert einwandfrei. 🥰

Kann geschlossen werden.