tbnobody/OpenDTU #1157: Inverters mostly unreachable

vo5tr0, 2023-07-17:
### What happened?

Since about 3 days my all my inverters went offline/ are unreachable by OpenDTU. I didn't change anything, didn't install any updates, didn't even touch the controller or its casing. It was running for weeks now without any interruption.

Today I was able to establish a connection again after restarting my inverters, restarting OpenDTU several times... but after 2 hours or so they were gone again :(

I also updated to the lastest firmware today without success...

How can I troubleshoot this issue? Is my controller dieing or something?

Controller: Olimex ESP32-POE-ISO

### To Reproduce Bug

No idea how to reproduce this. it just happens

### Expected Behavior

Connection to inverters

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

3b02ffe

### Relevant log/trace output

```Shell
Websocket: [/livedata][4] disconnect
21:17:35.234 > RX Period End
21:17:35.234 > All missing
21:17:35.234 > Nothing received, resend whole request
21:17:35.234 > TX ActivePowerControl Channel: 61 --> 51 80 40 41 86 80 14 25 04 81 0B 00 00 B4 00 01 86 80 DA 
21:17:37.220 > RX Period End
21:17:37.220 > All missing
21:17:37.220 > Nothing received, resend whole request
21:17:37.220 > TX ActivePowerControl Channel: 75 --> 51 80 40 41 86 80 14 25 04 81 0B 00 00 B4 00 01 86 80 DA 
21:17:39.248 > RX Period End
21:17:39.248 > All missing
21:17:39.248 > Nothing received, resend whole request
21:17:39.248 > TX ActivePowerControl Channel: 3 --> 51 80 40 41 86 80 14 25 04 81 0B 00 00 B4 00 01 86 80 DA 
21:17:41.281 > RX Period End
21:17:41.281 > All missing
21:17:41.281 > Nothing received, resend count exeeded
21:17:41.327 > TX ActivePowerControl Channel: 23 --> 51 81 10 12 29 80 14 25 04 81 0B 00 00 B4 00 01 86 80 77 
21:17:43.337 > RX Period End
21:17:43.337 > All missing
21:17:43.337 > Nothing received, resend whole request
21:17:43.337 > TX ActivePowerControl Channel: 40 --> 51 81 10 12 29 80 14 25 04 81 0B 00 00 B4 00 01 86 80 77 
21:17:45.364 > RX Period End
21:17:45.364 > All missing
21:17:45.364 > Nothing received, resend whole request
21:17:45.364 > TX ActivePowerControl Channel: 61 --> 51 81 10 12 29 80 14 25 04 81 0B 00 00 B4 00 01 86 80 77 
21:17:47.453 > RX Period End
21:17:47.453 > All missing
21:17:47.453 > Nothing received, resend whole request
21:17:47.453 > TX ActivePowerControl Channel: 75 --> 51 81 10 12 29 80 14 25 04 81 0B 00 00 B4 00 01 86 80 77 
21:17:49.500 > RX Period End
21:17:49.500 > All missing
21:17:49.500 > Nothing received, resend whole request
21:17:49.500 > TX ActivePowerControl Channel: 3 --> 51 81 10 12 29 80 14 25 04 81 0B 00 00 B4 00 01 86 80 77 
21:17:51.451 > RX Period End
21:17:51.451 > All missing
21:17:51.451 > Nothing received, resend count exeeded
21:17:51.482 > TX ActivePowerControl Channel: 23 --> 51 81 85 26 70 80 14 25 04 81 0B 00 00 B4 00 01 86 80 8F 
21:17:53.511 > RX Period End
21:17:53.511 > All missing
[... 659 lines cut ...]
21:21:09.590 > TX SystemConfigPara Channel: 40 --> 15 81 85 26 70 80 14 25 04 80 05 00 64 B5 94 9E 00 00 00 00 00 00 00 00 8B 3F 18 
21:21:09.788 > RX Period End
21:21:09.788 > All missing
21:21:09.788 > Nothing received, resend whole request
21:21:09.788 > TX SystemConfigPara Channel: 61 --> 15 81 85 26 70 80 14 25 04 80 05 00 64 B5 94 9E 00 00 00 00 00 00 00 00 8B 3F 18 
21:21:10.015 > RX Period End
21:21:10.015 > All missing
21:21:10.015 > Nothing received, resend whole request
21:21:10.015 > TX SystemConfigPara Channel: 75 --> 15 81 85 26 70 80 14 25 04 80 05 00 64 B5 94 9E 00 00 00 00 00 00 00 00 8B 3F 18 
21:21:10.245 > RX Period End
21:21:10.245 > All missing
21:21:10.245 > Nothing received, resend whole request
21:21:10.245 > TX SystemConfigPara Channel: 3 --> 15 81 85 26 70 80 14 25 04 80 05 00 64 B5 94 9E 00 00 00 00 00 00 00 00 8B 3F 18 
21:21:10.453 > RX Period End
21:21:10.453 > All missing
21:21:10.453 > Nothing received, resend count exeeded
21:21:10.479 > TX ActivePowerControl Channel: 23 --> 51 81 85 26 70 80 14 25 04 81 0B 00 03 E8 00 01 D0 40 46 
21:21:12.559 > RX Period End
21:21:12.559 > All missing
21:21:12.559 > Nothing received, resend whole request
21:21:12.559 > TX ActivePowerControl Channel: 40 --> 51 81 85 26 70 80 14 25 04 81 0B 00 03 E8 00 01 D0 40 46 
21:21:14.608 > RX Period End
21:21:14.608 > All missing
21:21:14.608 > Nothing received, resend whole request
```

jstammi, 2023-07-17:
Hi, just a guess as I faced similar situation end of last week. And in the end it seemed to have been caused by power adapter.
I had my ESP32 powered from usb of my fritzbox. And suddenly it stopped working (after months with no problems like this). It worked again after using a dedicated power adapter.

Your device makes me assume, that you are powering it by POE. If so, then I would suggest to try to power it by 5V directly ... ?

jstammi, 2023-07-18:
@Technikfan, please do not mix up things.
With #1080 we see receiving some data, stopping after some time.
Here it was fully working. Now it is not working at all.

@vo5tr0 I can reproduce the latter here. And at least with my setup (AzDelivery ESP32 D1 Mini / AZ-Delivery NRF24L01+) this does not look like a software issue. As with changing the power supply, it is fully working (1A power adapter -> usb hub -> openDTU) or never at all (using the usb port of the fritzbox sitting close to the openDTU). With keeping everything else the same.

(PS: before end of last week, since at least march, the openDTU worked like a charm using the fritzbox usb port power - whatever changed this, I cannot explain, yet)

I just did some BASIC measuring on it. With some result, that I fail, yet, to correlate to the symptoms.

I have some small man-in-the-middle usb device, showing me current and voltage consumed by attached usb devices. I checked with my voltmeter the shown voltage accuracy and it compared it shows low by 0,05V.

Connecting this between the usb hub and openDTU, it shows 4,44=4,49V/0,16A. Measuring the NRF24L01+ voltage by multimeter shows 3,29V. And openDTU is working nominally.

Connecting this between the fritzbox usb port and openDTU (only VCC+GND lines connected towards the ESP32), it shows 4,89=4,94V/0,16A, NRF24L01+ voltage by multimeter shows 3,30V. And openDTU does not receive any data from the inverter any longer.

Last weekend I puilled the openDTU to my office and used a different power supply and my openDTU was working nominally. Measuring this ones voltage with a different ESP32 connected, this shows 5,07V/0,15A.

So to me this does not look like being related to voltage level at all. But for sure - at least here for me - it is related somehow to the power supply.

As the ESP32 is working in all setups (web if reachable, could adjust settings always), I guess it has some effect on power transmitted from ESP32 to the NRF24L01+.

That makes me think, that changing the power supply could help.

Else, as mentioned by the troubleshooting section in the readme, there is a need for adding a capacitor close to NRF24L01+ VCC/GND connectors, if one increases TX power beyond it's default. Did you do so? Then please revert it. And if you then receive at least SOME data (indicated by some "Interrupt received" message(s) in the console), then you should add this capacitor definitely.

Else double check (measure) all wirings if some soldering or wire line are broken.

Else your radio maybe just gone and you should test if replacing it fixes your problem.

vo5tr0, 2023-07-18:
Ok, I just tried a USB power supply and it immediately worked. Then I unplugged my PoE switch for half an hour and tried again in PoE mode without the additional power supply which also was successful. So far everything looks good, I just don’t understand the problem. Maybe my switch didn’t provide enough current for some reason…

At least I know now where to look at and how to fix this issue.

Thank you!

JanD92, 2023-07-18:
I had exactly the same problem. It started yesterday with losing connection between the controller and inverter every now and then. Today nothing was working anymore. Changing power supplies and cables didn't help at all. I finally got it to work again by changing the frequency from 865.00 MHz (default) to 865,25 MHz. I don't know how the inverter decides which frequency it should use. Maybe it changes the frequency if something else is already using it like wifi routers do. Anyways it works for me for now.

I hope this helps other people that experience the same problem.

Controller: [ESP32 + CMT2300A](https://www.amazon.de/dp/B0C4B4GZCM)
Inverter: Hoymiles HMS-2000

tbnobody, 2023-07-18:
> I don't know how the inverter decides which frequency it should use.

The inverter always starts at 865MHz and has to be put on a target frequency by the DTU. That means that the Inverter operates at the frequency which you choose. If the inverter does not receive anything on it's target frequency for about 15min it switches back to 865MHz and waits again to be placed on a frequency of your choice. So in your case it might be that maybe one of your neighbors operates a device at 868,0MHz or something else happens on this frequency.

tbnobody, 2023-08-03:
I would close this issue as it is more less a hardware specific topic. Please feel free to reopen or create a discussion

k1llerk3ks, 2023-08-06:
I can confirm the solution of @tbnobody. with CMT2300A and HMS i had chosen 865Mhz initially. This brought me lots of disconnects for longer periods over the day. After changing it to 868Mhz everything works now very stable. Keep in mind, it takes a few minutes after connection is back after changing Frequency.