tbnobody/OpenDTU #1159: Add color initialization for multicolor OLED displays.

filo4444, 2023-07-18:
### Is your feature request related to a problem? Please describe.

I have got white/blue/yellow colors OLED SSD1306 display and display in OpenDTU works strange - colors of text looks like chosen randomly. Maybe is necessary to initialize display with set on only one color (white ?). Or should be used method of display library like SetDrawColor ?  

### Describe the solution you'd like

One color on all display.

tbnobody, 2023-07-19:
Multicolor displays are not supported at the moment. Will add a comment in the documentation.

StefanBruens, 2023-07-31:
For SSD1306 based "color" displays the colors are fixed, each area uses a different organic light emitting compoind. This is fixed at manufacturing time.

There are "real" color OLEDs, e.g. with an SSD1331 controller, where each pixel can have a distinct, software controllable color.

tbnobody, 2023-08-01:
Just also doublechecked the datasheet. SSD1306 only supports one color. This means there exists no display which supports any color change (as @StefanBruens wrote). Going to close this issue as there is nothing to do.

closed 2023-08-01 by tbnobody