tbnobody/OpenDTU #1169: Total Power versus Current Limit

nobbystyles, 2023-07-24:
### What happened?

I am trying to control the energy output of my 24-volt-batterie-system with an Hoymilesinverter HM-300 and the OpenDTU v23.7.22. In the process I experienced the energy output controlled by the current limit non-persistent in percent is at least not linear. My measuring instrument is a calibrated Tasmota-plug in conjunction with ioBroker. I also verified the differences with a "FRITZ!DECT200-plug". OpenDTU itself never shows more than 240 W "Total Power" even when it states in the inverter information "Current Limit: 300 W | 100 %"

Until 73 % it is quit linear - only a few watts above the reference (Tasmota-plug). Then there is a drop where the watts fall below reference.
Reaching 83 % OpenDTU indicates roundabout 240 watts and the reference has already reached 300 watts, max. performance of the HM-300.

### To Reproduce Bug

Just "Set Limit Non-Persistent" to 100 % on an HM-300 in the webinterface when the HM-300 is connected to a 24-volt-battery (26.5 volts in my case). "Total Power" will never show 300 watts although 300 watts are produced in fact.

### Expected Behavior

"Total Power" and "Current Limit" should match from 3% to 100 %.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.7.22.

### Anything else?

[image]

tbnobody, 2023-07-24:
OpenDTU has no influence on this. Only the percentage/absolute value is sent to the inverter. The control itself takes place exclusively in the inverter. However, you are welcome to check with an original DTU whether it behaves the same way there and log the transmission.

nobbystyles, 2023-07-24:
Ok - thank you for your explanation - so I have to deal with that. I do not feature an orginal DTU and therefore cannot help further.

closed 2023-07-24 by nobbystyles