tbnobody/OpenDTU #1174: HMS-400 not recognised properly

chris299, 2023-07-27:
### What happened?

I have a HMS-1600 and a HMS-400. The HMS-400 seems to be recognized incompletely, see screenshots.

[image]

[image]

[image]

### To Reproduce Bug

just use a HMS-400 ;-)

### Expected Behavior

showing the complete information in system information and the right absolute limit (400W)

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

23.7.22

tbnobody, 2023-07-27:
Could you please have a look whether this unknown value still occours tomorrow after a power cycle (at night) of the inverter? The values look all a little bit odd. That happens sometimes.

chris299, 2023-07-28:
unfortunately problem persists:
[image]

tbnobody, 2023-07-28:
Since the firmware build date also shows absolutely nonsensical values, something has probably changed in the data structure. 

To analyze this further you should look in the console for `DevInfoSimple` and `DevInfoAll` (will occur about every 15min) and accordingly the RX and TX lines for it and post the result.

chris299, 2023-07-28:
I don't see any "DevInfo". here a 20 min sample:

`Websocket: [/livedata][484] disconnect
19:10:01.744 > Fetch inverter: 112483748174
19:10:01.797 > TX RealTimeRunData 865.00 MHz --> 15 83 74 81 74 80 16 22 96 80 0B 00 64 C3 F6 6A 00 00 00 00 00 00 00 00 BC 10 29 
19:10:01.943 > Interrupt received
19:10:02.060 > RX 865.00 MHz --> 95 83 74 81 74 80 16 22 96 01 00 01 01 43 00 7D 01 93 00 00 23 E9 01 E1 09 44 7F | -72 dBm
19:10:02.166 > Interrupt received
19:10:02.298 > RX 865.00 MHz --> 95 83 74 81 74 80 16 22 96 82 13 88 01 7E 00 00 00 10 03 E8 01 30 00 01 16 2C 22 | -72 dBm
19:10:02.376 > RX Period End
19:10:02.376 > Success
19:10:06.746 > Fetch inverter: 116484675536
19:10:06.857 > TX RealTimeRunData 865.00 MHz --> 15 84 67 55 36 80 16 22 96 80 0B 00 64 C3 F6 6F 00 00 00 00 00 00 00 00 EC 2F C1 
19:10:06.916 > Interrupt received
19:10:07.073 > RX 865.00 MHz --> 95 84 67 55 36 80 16 22 96 01 00 01 01 79 01 46 00 27 00 49 00 94 00 EF 00 00 1D | -37 dBm
19:10:07.166 > Interrupt received
19:10:07.363 > RX 865.00 MHz --> 95 84 67 55 36 80 16 22 96 03 00 46 00 00 00 E4 00 00 01 70 00 00 42 D4 00 00 71 | -37 dBm
19:10:07.365 > Interrupt received
19:10:07.398 > RX 865.00 MHz --> 95 84 67 55 36 80 16 22 96 04 01 DD 09 2B 13 88 02 49 00 00 00 18 03 E8 00 E5 0B | -37 dBm
19:10:07.577 > Interrupt received
19:10:07.634 > RX 865.00 MHz --> 95 84 67 55 36 80 16 22 96 85 00 03 15 6C C8 | -37 dBm
19:10:07.790 > RX Period End
19:10:07.790 > Middle missing
19:10:07.790 > Request retransmit: 2
19:10:07.790 > TX RequestFrame 865.00 MHz --> 15 84 67 55 36 80 16 22 96 82 35 
19:10:07.886 > Interrupt received
19:10:07.953 > RX 865.00 MHz --> 95 84 67 55 36 80 16 22 96 02 06 43 00 00 41 4F 00 BA 01 F3 00 05 01 42 00 00 70 | -37 dBm
19:10:08.015 > RX Period End
19:10:08.015 > Success
19:10:08.445 > Websocket: [/livedata][485] connect
19:10:11.747 > Fetch inverter: 112483748174
19:10:11.877 > TX RealTimeRunData 865.00 MHz --> 15 83 74 81 74 80 16 22 96 80 0B 00 64 C3 F6 74 00 00 00 00 00 00 00 00 1C 91 16 
19:10:11.979 > Interrupt received
19:10:12.032 > RX 865.00 MHz --> 95 83 74 81 74 80 16 22 96 01 00 01 01 43 00 7A 01 8A 00 00 23 E9 01 E1 09 3F 1A | -72 dBm
19:10:12.180 > Interrupt received
19:10:12.350 > RX 865.00 MHz --> 95 83 74 81 74 80 16 22 96 82 13 88 01 76 00 00 00 10 03 E8 01 30 00 01 0D F0 ED | -72 dBm
19:10:12.386 > RX Period End
19:10:12.386 > Success
19:10:16.750 > Fetch inverter: 116484675536
19:10:16.807 > TX RealTimeRunData 865.00 MHz --> 15 84 67 55 36 80 16 22 96 80 0B 00 64 C3 F6 79 00 00 00 00 00 00 00 00 8C C9 51 
19:10:16.892 > Interrupt received
19:10:16.962 > RX 865.00 MHz --> 95 84 67 55 36 80 16 22 96 01 00 01 01 7A 01 46 00 26 00 48 00 92 00 EC 00 00 1B | -37 dBm
19:10:17.029 > Interrupt received
19:10:17.214 > RX 865.00 MHz --> 95 84 67 55 36 80 16 22 96 02 06 43 00 00 41 4F 00 BA 01 F3 00 05 01 42 00 00 70 | -37 dBm
19:10:17.426 > Interrupt received
19:10:17.428 > RX 865.00 MHz --> 95 84 67 55 36 80 16 22 96 03 00 45 00 00 00 E1 00 00 01 70 00 00 42 D4 00 00 77 | -37 dBm
19:10:17.464 > Interrupt received
19:10:17.612 > RX 865.00 MHz --> 95 84 67 55 36 80 16 22 96 04 01 DD 09 2A 13 88 02 42 00 00 00 18 03 E8 00 E5 01 | -37 dBm
19:10:17.669 > Interrupt received
19:10:17.818 > RX 865.00 MHz --> 95 84 67 55 36 80 16 22 96 85 00 03 82 04 37 | -37 dBm
19:10:17.874 > RX Period End
19:10:17.874 > Success
19:10:21.758 > Fetch inverter: 112483748174
19:10:21.826 > TX RealTimeRunData 865.00 MHz --> 15 83 74 81 74 80 16 22 96 80 0B 00 64 C3 F6 7E 00 00 00 00 00 00 00 00 BC EF C2 
19:10:21.913 > Interrupt received
19:10:22.017 > RX 865.00 MHz --> 95 83 74 81 74 80 16 22 96 01 00 01 01 43 00 7E 01 95 00 00 23 E9 01 E1 09 41 7F | -72 dBm
19:10:22.074 > Interrupt received
19:10:22.337 > RX 865.00 MHz --> 95 83 74 81 74 80 16 22 96 82 13 88 01 80 00 00 00 10 03 E8 01 2F 00 01 F9 BB BB | -72 dBm
19:10:22.458 > Websocket: [/livedata][485] disconnect
19:10:22.460 > RX Period End
19:10:22.460 > Success
19:10:22.606 > Websocket: [/livedata][486] connect
19:10:26.749 > Fetch inverter: 116484675536
19:10:26.829 > TX RealTimeRunData 865.00 MHz --> 15 84 67 55 36 80 16 22 96 80 0B 00 64 C3 F6 83 00 00 00 00 00 00 00 00 28 B3 75 
19:10:26.929 > Interrupt received
19:10:27.012 > RX 865.00 MHz --> 95 84 67 55 36 80 16 22 96 01 00 01 01 7E 01 47 00 2E 00 4D 00 B0 00 FD 00 00 20 | -37 dBm
19:10:27.146 > Interrupt received
19:10:27.378 > RX 865.00 MHz --> 95 84 67 55 36 80 16 22 96 02 06 43 00 00 41 50 00 BA 01 F4 00 05 01 42 00 00 68 | -37 dBm
19:10:27.524 > Interrupt received
19:10:27.532 > RX 865.00 MHz --> 95 84 67 55 36 80 16 22 96 03 00 4B 00 00 00 F2 00 00 01 70 00 00 42 D4 00 00 6A | -37 dBm
[... 2731 lines cut ...]
19:31:12.935 > RX Period End
19:31:12.935 > Success
19:31:17.380 > Fetch inverter: 116484675536
19:31:17.436 > TX RealTimeRunData 865.00 MHz --> 15 84 67 55 36 80 16 22 96 80 0B 00 64 C3 FB 65 00 00 00 00 00 00 00 00 B6 C0 73 
19:31:17.537 > Interrupt received
19:31:17.635 > RX 865.00 MHz --> 95 84 67 55 36 80 16 22 96 01 00 01 01 96 01 53 01 1B 00 D8 04 81 02 E1 00 00 56 | -39 dBm
19:31:17.691 > Interrupt received
19:31:17.884 > RX 865.00 MHz --> 95 84 67 55 36 80 16 22 96 02 06 4C 00 00 41 59 00 C3 01 FD 00 05 01 50 00 00 0C | -39 dBm
19:31:18.023 > Interrupt received
19:31:18.079 > RX 865.00 MHz --> 95 84 67 55 36 80 16 22 96 03 00 D0 00 00 02 BD 00 00 01 70 00 00 42 DC 00 00 B4 | -39 dBm
19:31:18.250 > Interrupt received
[... cut ...]
19:31:22.598 > TX RealTimeRunData 865.00 MHz --> 15 83 74 81 74 80 16 22 96 80 0B 00 64 C3 FB 6A 00 00 00 00 00 00 00 00 46 81 4F 
19:31:22.697 > Interrupt received
19:31:22.774 > RX 865.00 MHz --> 95 83 74 81 74 80 16 22 96 01 00 01 01 56 02 98 08 DA 00 00 23 FD 01 F5 09 4B C2 | -72 dBm
19:31:22.960 > Interrupt received
19:31:23.063 > RX 865.00 MHz --> 95 83 74 81 74 80 16 22 96 82 13 87 08 64 00 00 00 5A 03 E8 01 22 00 01 C9 55 C0 | -71 dBm
19:31:23.114 > RX Period End
19:31:23.114 > Success
`

chris299, 2023-07-29:
after another power cycle it works today.... strange...
[image]

closed 2023-07-29 by chris299