tbnobody/OpenDTU #1187: Spurious bad metrics readings

easimon, 2023-07-31:
### What happened?

In the past I had two spurious issues with bad metrics. Since total yield (AC) is what I look at most, this is where I found it.

* A single sample that is totally off (in one reading, total yield jumps from 300KWh to 18000KWh and with the following reading, it's back to 300). Don't know what the concrete values were here.
* A series of samples for total yield after night that are lower than the last reading before night: Concrete example see below. The last value before night (which is then kept for the whole night) was higher than the first readings in the morning. 

[image]

I suspect non-atomic updates of the variables to be the root cause (prometheus endpoint reading from variables that are in the middle of an update during read). Is this possible, and if so, something that can be fixed?

Why do I care? I would not, if these errors were only temporary -- but in prometheus they are kept like forever, and errors like these are quite catastrophic, because prometheus expects counters to monotonously increase... and every decrease is treated as a "reset to 0 and then again increase to the value". And since prometheus does not allow for correction/deletion of single samples, this manifests forever (until the end of retention, for me a year). The only safe way to get rid of these errors is "delete your complete history for that metric, or revert to a backup that did not contain the bad samples, and accept the gap". I did the latter twice.

For the first case (300 -> 18000 -> 300), the total yield calculated by prometheus is 18300, for the second case (452.997 -> 452.996 -> 452.997) the result is 905.994... doubling the measured yield.

### To Reproduce Bug

I actually don't know -- maybe hammer the prometheus endpoint for a long time and watch out for implausible readings? 

### Expected Behavior

Counter metrics must not report wrong readings. A counter may only increase -- except for real resets, like the daily yield that resets to 0 every day.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.7.10

tbnobody, 2023-07-31:
I've found a location where a non-atomic operation could occour. Will doublecheck tomorrow and come back to you in the evening if no side effects occoured during the day

tbnobody, 2023-08-01:
Version v23.8.1 will be available in some minutes. Please doublecheck whether the issue still occours.

easimon, 2023-08-02:
Will do. But don't expect feedback too soon. I am using OpenDTU for 5 months now, and the issue occured only twice :)

tbnobody, 2023-08-02:
Then I would close this issue for now. Please feel free to open it again (or create a new one) if the issue still occours.

closed 2023-08-02 by tbnobody

easimon, 2023-08-09:
It happened again - the YieldTotal counters decreased by just a little bit. 
And this time, it occured for the AC counter as well as for *one* DC counter, at the very same time.

The values are extremely close to each other (and the same values repeat before/after the decrease), so I guess that is the smallest difference that can be represented in whatever the counter type in the inverter is.

It is not single samples that are wrong, but the DTU reads the same values a few times -- which makes timing issues in non-atomic variable access quite improbable. What I also noticed: this seems to happen over night. The wrong decrease happens immediately in the morning, with the first few readings.

I could think of two reasons for this: 

* The inverter sometimes "forgets" to store the latest values before shutdown over night, and starts with slightly outdated/smaller counters in the morning. 

* It's some conversion/rounding error when reading. In the code, I've seen there is a `byteAssign_t` definition, that also holds `uint8_t digits; // number of valid digits after the decimal point`. For the YieldTotal counters, the digits property is `3`. Not sure what that means though.

The prometheus endpoint prints the numbers as `%f`, which has 6 decimal digits, e.g.

```
opendtu_YieldTotal{serial="...",unit="1",name="...",type="AC",channel="0"} 463.846008
```

At least in the examples below, the decrease/error is **after** the 3rd digit (but only when cutting off after 3 digits! When rounding to 3 digits, the error would also round up, so `%.3f` would most probably not help.

Do you have any idea what's happening here -- and also if that's something OpenDTU could fix/mitigate?

Below are screenshots of the affected counters, and the "last update" counter for the same time frame.
 
## AC: 457.419006 -> 457.419983 -> 457.419006 -> 457.421021
[image]

## DC channel 0: 241.350006 - > 241.350998 ->  241.350006 -> 241.350998
[image]

## DC channel 1: no decrease
[image]

## Last update
[image]