tbnobody/OpenDTU #1190: [Request] Frequency preset for another country

jomp16, 2023-08-01:
### Is your feature request related to a problem? Please describe.

Yes, since the defaults (865 Mhz) is way out of range of working frequencies on Brazil, making the inverters non reachable.

When I changed to 915.75 MHz - 916.00 MHz (maybe the existing factory Hoymiles DTU is making the frequency changes since sometimes it's unreachable and then reachable.), the OpenDTU can connect and pull data from two of the Hoymiles inverters that I have installed.

### Describe the solution you'd like

Adding a frequency preset list (aka: Germany 863 Mhz to 870 Mhz, and Brazil from 915 Mhz to 928 Mhz (checked on Anatel, our Brazilian version of FCC)).

Also removing the warning message of "frequency out of EU range", since I'm on Brazil and Anatel has certified 915 – 928 MHz as a working frequencies of the DTU dongle.

### Describe alternatives you've considered

At least removing the warning about being out of EU range (something about "I accept the risks" and then not showing anymore?).

### Additional context

[image]
---
[image]
---
[Anatel certification of the Hoymiles DTU Lite S with the working frequencies - NCC_19719_23_CH_International.pdf](https://github.com/tbnobody/OpenDTU/files/12228411/NCC_19719_23_CH_International.pdf)

tbnobody, 2023-08-01:
That's interesting that it's working because the initial packet is always sent on 868MHz... That's hard coded in the inverter. On this frequency a packet is sent to move the communication to another frequency.

jomp16, 2023-08-01:
Well, maybe the frequency of the inverters on Europe (probably where you do live?) is always set at 868MHz, but the inverters on Brazil, maybe it's a different variation from ones sent to Europe and the default frequency is set at another one?

gpa18mx, 2023-09-11:
In Mexico is also 915

Fribur, 2023-11-27:
FCC gave approval for [905-925 MHz in North America.](https://fcc.report/company/Hoymiles-Power-Electronics-Inc ). I cannot imagine HMS units sold in US (e.g. hms-350-1t-na) try first 868 Mhz as this would be in violation to FCC 15C. How can I find out which frequency was hard coded for the first packet into the Nort American units? I am unable to find any frequency between 905 and 925 to communicate with the inverter. I fact I can only go up to 923.5 in opendtu. Now I am unsure now if the cmt2300a might be defect or I am just not waiting long enough on any of the tested frequencies.

tbnobody, 2024-01-13:
Whould close this issue because the current development is discussed in #1540

closed 2024-01-13 by tbnobody