tbnobody/OpenDTU #1194: Homeassistant MQTT: neue Namenrichtlinie

smoki3, 2023-08-02:
### What happened?

Hallo,

seit dem aktuellen Update von Home Assistant bekommt man die Meldung, dass die übertragenen MQTT Namen nicht mehr zulässig sind und aktuell noch ein Workaround betrieben wird.

[image]

### To Reproduce Bug

Update Homeassistant auf 2023.08

### Expected Behavior

kompatibilität mit Homeassistant

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.8.1

knurrdog, 2023-08-02:
[The naming of MQTT entities changes to correspond with HA guidelines/](https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/)

LennartF22, 2023-08-05:
Sollte kein großes Problem sein, die Namen in der Auto-Discovery-Config anzupassen. Falls noch keiner dran ist, könnte ich mich darum kümmern und eine PR stellen (dann am besten wahrscheinlich erstmal nur als Opt-in bei den MQTT-Einstellungen, damit es bei alten Home Assistant Versionen nichts durcheinander bringt).

heilingbrunner, 2023-08-05:
Könnt Ihr das am Ende etwas detailierter erklären (welches Gerät, welches Menü, etc.), damit man das nachvollziehen kann ?
Laut HA soll man diese Naming-Convention Warnung ignorieren und alles geht wie vorher, aber bei mir sind alle Werte 'unavailable'.
Danke.

Norm71, 2023-08-05:
Wenn der Wechselrichter abschaltet sind die Werte nicht verfügbar. Soblad er produziert sollten die Werte wieder zu sehen sein. So ist es zumindest bei mir

heilingbrunner, 2023-08-05:
Ja, dass nachts keine Werte kommen ist bekannt. Aber ich sehe im openDTU Web-UI, dass das Gerät funktioniert. Hat ja auch wochenlang funktioniert in Home-Assistant. Der 'binary_sensor.hm800_producing' ist hingegen immer noch verfügbar. Alle 'sensor.hm800_...' Entities nicht. Die wurden in der Warnung ja auch moniert. Hm ...

LennartF22, 2023-08-05:
Habe in den MQTT-Einstellungen eine Option eingebaut, um auf das neue Namensschema zu wechseln. Falls es jemand ausprobieren möchte: https://github.com/LennartF22/OpenDTU/actions/runs/5773100117

Kann natürlich nicht sagen, ob und wann das in dieser Form gemerged wird.

LeoWinterDE, 2023-08-05:
Rel. https://github.com/lumapu/ahoy/issues/1066

sholvatealc, 2023-08-06:
> Habe in den MQTT-Einstellungen eine Option eingebaut, um auf das neue Namensschema zu wechseln. Falls es jemand ausprobieren möchte: https://github.com/LennartF22/OpenDTU/actions/runs/5773100117
> 
> Kann natürlich nicht sagen, ob und wann das in dieser Form gemerged wird.

Hab des jetzt mal angewendet. Muss ich im Homeassistant dann auch noch eine Aktion ausführen?

sarmbruster, 2023-08-06:
Vielen Dank @LennartF22 . Ich habe es bei mir installiert und in der UI `Altes Namensschema` abgeschaltet. In HA ( 2023.8.1)  ist die oben beschriebene Fehlermeldung weg.
Daher strong +1 fürs mergen.

sholvatealc, 2023-08-08:
Nachtrag:
Jetzt ist zwar der Fehler weg. Aber unter MQTT Homeassistent ist kommen keine Parameter mehr an die ich für Ertrag auswählen kann. Bin wieder auf alte Namen gegangen. Jetzt geht es wieder.

wesseli01, 2023-08-11:
Vielen Dank @LennartF22. Ebenso in der UI das alte Namensschema deaktiviert. Alle Entitaeten sind nun sauber angelegt worden. Die Funktion sollte im neuen Release integriert werden.

morph027, 2023-08-11:
Auch von mir ein 👍 , läuft.

Omega13x, 2023-08-11:
Hallo.
Ich habe die Version auch mal installiert und die alten Namensrichtlinien deaktiviert. Ich habe OpenDTU und HA komplett neu gestartet. Ich kann aber keine Änderung feststellen. Habe ich irgendetwas vergessen?

wesseli01, 2023-08-11:
@Omega13x und was sehen Sie unter den Integrationseinträgen von MQTT?

wesseli01, 2023-08-11:
@LennartF22 - Könnten Sie die Änderung in die neue Version 23.8.8 integrieren?

LennartF22, 2023-08-11:
@wesseli01 Erledigt, sollte gleich hier verfügbar sein: https://github.com/LennartF22/OpenDTU/actions/runs/5836940847

wesseli01, 2023-08-12:
@LennartF22 , Vielen vielen Dank. Das hilft uns sehr.

Phil1988, 2023-08-20:
Wie ist der aktuelle Stand? Wird das in die offizielle Version hier integriert?
Ich habe es nicht von Lennarts Version getestet, weil ich hier die originale Version nehmen möchte und später dann das FW Update durchführen möchte, sobald es integriert ist.
Danke @LennartF22 übrigens!

sholvatealc, 2023-08-24:
@LennartF22
Hab jetzt noch 2 Module an meinen Wechselrichter ran. Jetzt kommt die Meldung in Ha.
MQTT entity name starts with the device name in your config {'name': 'Balkonkraftwerk_01 CH4 Power', 'state_topic': 'solar/116185129399/4/power', 'unique_id': '116185129399_ch4_Power', 'unit_of_measurement': 'W', 'device': {'name': 'Balkonkraftwerk_01', 'identifiers': ['116185129399'], 'configuration_url': 'http://192.168.178.91', 'manufacturer': 'OpenDTU', 'model': 'HM-1000, HM-1200, HM-1500', 'sw_version': 'v23.8.22', 'connections': []}, 'expire_after': 15, 'device_class': <SensorDeviceClass.POWER: 'power'>, 'state_class': <SensorStateClass.MEASUREMENT: 'measurement'>, 'availability_mode': 'latest', 'payload_not_available': 'offline', 'payload_available': 'online', 'enabled_by_default': True, 'force_update': False, 'qos': 0, 'encoding': 'utf-8'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'CH4 Power'

Ist das ein Fehler von HA oder Open-DTU. 23.8.22 ist bei mit drauf

Gitsarry, 2023-08-27:
The OP warning disappeared for me after update to HA 2023.8.4.
Matches this point of 2023.8.4 changelog:
* Remove repair issue for MQTT discovered items https://github.com/home-assistant/core/pull/98768

sholvatealc, 2023-08-30:
> ich bin auf Version 23.8.28. Wo finde ich den die Option zum Umstellen und wie heißt sie genau?

Die ist in einer ich sag jetzt mal "privaten" Änderung von @LennartF22 hier https://github.com/LennartF22/OpenDTU/actions/runs/5836940847

Dann bei Einstellung/MQTT ganz unten.

MarpleA, 2023-08-31:
ah ok. danke. ich dachte das wäre schon in das offizielle Release eingefloßen.

Heiko-L, 2023-09-10:
> The OP warning disappeared for me after update to HA 2023.8.4. Matches this point of 2023.8.4 changelog:
> 
>     * Remove repair issue for MQTT discovered items [Remove repair issue for MQTT discovered items home-assistant/core#98768](https://github.com/home-assistant/core/pull/98768)

That is no solutuion to the problem

sholvatealc, 2023-09-10:
@Heiko-L 
In the meantime, use this version until it is incorporated into the official one.
Then uncheck “Old naming scheme” at the bottom under Settings/MQTT.
https://github.com/LennartF22/OpenDTU/actions/runs/5836940847

sholvatealc, 2023-10-03:
@LennartF22
Magst Du das aktuelle Built mit Deinem Namenschalter reinstellen ? Oder ist des jetzt schon abgehandelt das Thema?
Grüße Markus

LennartF22, 2023-10-05:
@sholvatealc Ich könnte meine Änderungen zwar nochmal mit der aktuellen OpenDTU-Version zusammenführen, aber das ist ja irgendwie nicht Sinn der Sache. Stattdessen müsste entweder die PR von @tbnobody gemerged werden oder das Problem anderweitig angegangen werden. Nur weil die Meldung nicht mehr ganz so „in your Face“ ist, besteht das unterliegende Problem ja weiterhin. Spätestens wenn Home Assistant den Legacy-Mode in ein paar Monaten entfernt, muss ja irgendetwas passieren (wenn bei den Leuten nicht überall der Wechselrichtername doppelt in den Entitäten drin stehen soll).

S-Przybylski, 2023-10-13:
Dear @tbnobody could you have a look on this please? 
Each restart of Home Assistant my protocol is flodded with warnings. Thanks

TheDK, 2023-11-14:
+1 - this should be changed as it is now HA standard and will hit every new user...

tbnobody, 2023-11-15:
Implemented in 71d1b3b84656e9ee477f3839d9fcb33741ec12ac

closed 2023-11-15 by tbnobody