tbnobody/OpenDTU #1198: Calculated Limit 210W

Bankaifan, 2023-08-02:
### What happened?

Nach dem Update auf Version 23.8.1 zeigt die Oberfläche für meine HM-300 das aktuelle Limit von 100% / 210W an.
[image]

[image]

### To Reproduce Bug

-

### Expected Behavior

Das Limit sollten wohl 300W sein

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

77528f6

### Relevant log/trace output

```Shell
Interrupt received
14:55:18.551 > RX Channel: 75 --> 95 83 21 85 95 83 21 85 95 82 13 89 01 4A 00 00 00 0E 03 E8 00 EF 00 01 25 45 AD | -80 dBm
14:55:18.888 > RX Period End
14:55:18.888 > Success
14:55:19.230 > Websocket: [/livedata][7] disconnect
14:55:22.472 > Fetch inverter: 112182853055
14:55:22.659 > TX RealTimeRunData Channel: 61 --> 15 82 85 30 55 80 19 47 92 80 0B 00 64 CA 52 3A 00 00 00 00 00 00 00 00 74 85 87 
14:55:22.859 > Interrupt received
14:55:23.058 > RX Channel: 3 --> 95 82 85 30 55 82 85 30 55 01 00 01 01 2E 00 73 01 5B 00 00 05 E1 01 9F 08 FB 1A | -80 dBm
14:55:23.299 > Interrupt received
14:55:23.484 > RX Channel: 3 --> 95 82 85 30 55 82 85 30 55 82 13 89 01 4C 00 00 00 0E 03 E8 00 DD 00 03 4E 37 82 | -80 dBm
14:55:23.669 > RX Period End
14:55:23.670 > Success
14:55:27.829 > Fetch inverter: 112183823895
14:55:28.054 > TX RealTimeRunData Channel: 75 --> 15 83 82 38 95 80 19 47 92 80 0B 00 64 CA 52 3F 00 00 00 00 00 00 00 00 24 BA 23 
14:55:28.177 > Interrupt received
14:55:28.279 > RX Channel: 23 --> 95 83 82 38 95 83 82 38 95 82 13 89 01 4F 00 00 00 0F 03 E8 00 E5 00 01 2E B6 5B | -80 dBm
14:55:28.490 > RX Period End
14:55:28.490 > Middle missing
14:55:28.490 > Request retransmit: 1
14:55:28.490 > TX RequestFrame Channel: 3 --> 15 83 82 38 95 80 19 47 92 81 74 
14:55:28.690 > RX Period End
14:55:28.690 > Middle missing
14:55:28.690 > Request retransmit: 1
14:55:28.690 > TX RequestFrame Channel: 23 --> 15 83 82 38 95 80 19 47 92 81 74 
14:55:28.892 > Interrupt received
14:55:28.995 > RX Channel: 61 --> 95 83 82 38 95 83 82 38 95 01 00 01 01 6B 00 61 01 5E 00 03 99 AC 00 02 08 F7 0A | -80 dBm
14:55:29.099 > RX Period End
14:55:29.099 > Success
14:55:32.956 > Fetch inverter: 112183213583
14:55:33.098 > TX RealTimeRunData Channel: 40 --> 15 83 21 35 83 80 19 47 92 80 0B 00 64 CA 52 44 00 00 00 00 00 00 00 00 D6 AC 04 
14:55:33.544 > RX Period End
14:55:33.544 > All missing
14:55:33.544 > Nothing received, resend whole request
14:55:33.544 > TX RealTimeRunData Channel: 61 --> 15 83 21 35 83 80 19 47 92 80 0B 00 64 CA 52 44 00 00 00 00 00 00 00 00 D6 AC 04 
14:55:33.705 > Interrupt received
14:55:33.825 > RX Channel: 3 --> 95 83 21 35 83 83 21 35 83 82 13 88 01 4F 00 00 00 0F 03 E8 00 DB 01 33 30 96 69 | -80 dBm
14:55:34.078 > RX Period End
14:55:34.078 > Middle missing
14:55:34.078 > Request retransmit: 1
14:55:34.078 > TX RequestFrame Channel: 75 --> 15 83 21 35 83 80 19 47 92 81 CC 
14:55:34.219 > RX Period End
14:55:34.219 > Middle missing
14:55:34.219 > Request retransmit: 1
14:55:34.219 > TX RequestFrame Channel: 3 --> 15 83 21 35 83 80 19 47 92 81 CC 
14:55:34.355 > RX Period End
14:55:34.355 > Middle missing
14:55:34.355 > Request retransmit: 1
14:55:34.355 > TX RequestFrame Channel: 23 --> 15 83 21 35 83 80 19 47 92 81 CC 
14:55:34.531 > RX Period End
14:55:34.531 > Middle missing
14:55:34.531 > Request retransmit: 1
14:55:34.531 > TX RequestFrame Channel: 40 --> 15 83 21 35 83 80 19 47 92 81 CC 
14:55:34.732 > RX Period End
14:55:34.732 > Middle missing
14:55:34.732 > Request retransmit: 1
14:55:34.732 > TX RequestFrame Channel: 61 --> 15 83 21 35 83 80 19 47 92 81 CC 
14:55:34.954 > RX Period End
14:55:34.954 > Middle missing
14:55:34.954 > Retransmit timeout
14:55:35.148 > TX AlarmData Channel: 75 --> 15 83 21 35 83 80 19 47 92 80 11 00 64 CA 52 44 00 00 00 00 00 00 00 00 0C B7 DF
```

tbnobody, 2023-08-02:
This change was introduced on October 20th 2022 
https://github.com/tbnobody/OpenDTU/blob/77528f6e6a18ecc12069d8250add489d742746ac/lib/Hoymiles/src/parser/DevInfoParser.cpp#L29C16-L29C16

Based on your hardware part number it seems that you have a limited version of the inverter. This was at least the case when talking to other people how had the same part number. Are you certain that your inverters can deliver the 300W? Did you measure it using a power meter or something else? 

Try setting a temporary limit to 100W. What will be the display of the limit after 1 minute and after 15 minutes?

Bankaifan, 2023-08-02:
The fourth inverter was installed yesterday. 
The others are running for several month now. Each with a peak of 306W measured with a Shelly 3EM (Household-Powerusage ~150W and feeding to the grid ~770W). So yes - all of them deliver the expected ~300Ws

tbnobody, 2023-08-02:
Which version did you run previously? 

> Try setting a temporary limit to 100W. What will be the display of the limit after 1 minute and after 15 minutes?

What was the result?

Bankaifan, 2023-08-02:
Quite interesting, i have a screenshot of my DTU form the 11. April, where the limit says 300W | 100%. 

Prev. Version should have been #e8bdaf9198

Set all of them to 100W - says now 100W | 48%
[image]

Waiting now 15 mins.

Bankaifan, 2023-08-02:
Oh wait... right after i've send the last message i now see this:
[image]

Bankaifan, 2023-08-02:
No change after 15 minutes. Set it back to 100% - but still 210W limit

This was an 11. April with the latest version at this time.
[image]

tbnobody, 2023-08-02:
Ok, just as a expalantion... 100W was sent to the inverter as limit. Just right after sending the value, the percentage is calculated on the ESP by dividing the 100 / 210 = 48%
After several minutes the inverter is polled for the current limit. The inverter just returns the percent value. In your case 33%. 
100W / 33% = 300W
So 300W is really the correct value.

Question now is why it showed 300W previously and now 210W. 

Do all have the same Hardware Part Number?
Could you try restarting the DTU? The device information is only fetched one times for each inverter after the startup of the DTU.

Bankaifan, 2023-08-02:
I restarted the DTU, but still shows 210W.
One Inverter has a different Hardware part number, but it's not the new one and I saw 1100W yesterday, so should not be a problem related to the newest inverter.

Bankaifan, 2023-08-02:
Okay. I've downgraded to https://github.com/tbnobody/OpenDTU/releases/tag/v23.7.12

Now it shows 300W again. For all 4 inverters.
23.7.12 was my previous version, not 23.7.0 I mentioned above

Bankaifan, 2023-08-02:
I can give you a public ip-address if you want to check/try something yourself

tbnobody, 2023-08-02:
ahhhhhhhhhh Sorry! Habs grad gesehen.... Gibt im laufe des Abends einen Fix

TommyRhb, 2023-08-02:
Kein Problem :) Danke! :)

commit 686112e4 (2023-08-02): Fix: Wrong detection of HM-300 inverters

Was introduced in v23.8.1; Fixes #1198