tbnobody/OpenDTU #1199: Please insert a timestamp in the restful API

Frogowitsch, 2023-08-02:
### Is your feature request related to a problem? Please describe.

Its hard to synchronize with other devices. It would also be nice to have the ability to set a delta time for ntp sync.

### Describe the solution you'd like

A unix timestamp in the restful api would be rather easy to implement and help a lot.

Frogowitsch, 2023-08-02:
I know, that there is a possibility to access the /api/ntp/time , but this is a second access and i use a slow 8266 to read the values, so if I read the /api/ntp/time for time and then read /api/livedata/status for the data, I have a delay due to 2 accesses (my Wifi is rather bad and the 8266 is also slow).

stefan123t, 2024-08-05:
@Frogowitsch I understand and second your request for a timestamp in several of the API endpoints.

This would need to be considered to in our REST API documentation #378.

---

I have also seen that the dusk / dawn times have to be queried under Info > NTP whereas the setting for which type of sunset / dawn can be set under Settings > NTP only. 
In order to see the effect of the four sunset type settings, one has to save the setting and switch to the other tab. This could be more convenient in the same dialog.

Settings > NTP

NTP Settings 
Location Configuration
Sunset type:
* Standard dawn (90.8°)
* Nautical dawn (102°)
* Civil dawn (96°) *default*
* Astronomical dawn (108°)

Why is the list not sorted by degrees ° and the IMHO Civil dawn (96°) marked as default ?

Settings > NTP

NTP Information
Current Time
Sunrise	Monday, August 05 2024 05:24:00
Sunset	Monday, August 05 2024 21:26:00

stefan123t, 2024-10-18:
@tbnobody I would add to the request that we add a Leaflet Geosearch address entry field to the Settings > NTP Settings.

Look at this Stackoverflow for some example implementations using Leaflet and pure Java Script:

**get latitude & longitude as per address given for leaflet**
https://stackoverflow.com/questions/15919227/get-latitude-longitude-as-per-address-given-for-leaflet

stefan123t, 2025-05-23:
Here is a rather short Java Script code snippet needed to provide a Leaflet Geosearch address entry field.

After entering your address and hitting search it will do a Geolocation query to find suitable POIs matching your search query over the Nominatim OpenStreetmap API.
You can click on the POIs in the list and the Leaflet Map display will refocus on the given POI and update the Latitude and Longitude values in the DOM model.

@tbnobody @schlimmchen I do not know VUE well enough in order to integrate it into OpenDTU's Settings > NTP Configuration page.

[image]

<details><summary>Java Script Details for Leaflet Map Display and OpenStreetmap Nominatim API lookup</summary>
<p>

```javascript
<html>
  <head>
    <title>Leaflet Address Lookup and Coordinates</title>
    <meta charset="utf-8">

    <link rel="stylesheet" href="https://unpkg.com/leaflet@1.9.4/dist/leaflet.css" integrity="sha256-p4NxAoJBhIIN+hmNHrzRCf9tD/miZyoHS5obTRR9BMY=" crossorigin=""/>
    <script src="https://unpkg.com/leaflet@1.9.4/dist/leaflet.js" integrity="sha256-20nQCchB9co0qIjJZRGuk2/Z9VM+kNiyxNV1lvTlZBo=" crossorigin=""></script>

    <style type="text/css">
    html,
    body {
      width: 100%;
      padding: 0;
      margin: 0;
    }

    .container {
      width: 95%;
      max-width: 980px;
      padding: 1% 2%;
      margin: 0 auto
    }

    #lat,
    #lon {
      text-align: right
    }

    #map {
      width: 100%;
      height: 50%;
      padding: 0;
      margin: 0;
    }

    .address {
      cursor: pointer
    }

    .address:hover {
      color: #AA0000;
      text-decoration: underline
    }
    </style>
  </head>

  <body>
    <div class="container">
      <b>Coordinates</b>
      <form>
        <input type="text" name="lat" id="lat" size=12 value="">
        <input type="text" name="lon" id="lon" size=12 value="">
      </form>
      <b>Address Lookup</b>
      <div id="search">
        <input type="text" name="addr" value="" id="addr" size="58" />
        <button type="button" onclick="addr_search();">Search</button>
        <div id="results"></div>
      </div>

      <br />
      <div id="map"></div>
    </div>

    <script type="text/javascript">
    // Berlin
    var startlat = 52.51088500;
    var startlon = 13.39893670;

    var options = {
      center: [startlat, startlon],
      zoom: 7
    }

    document.getElementById('lat').value = startlat;
    document.getElementById('lon').value = startlon;

    var map = L.map('map', options);
    var nzoom = 7;

    L.tileLayer('http://{s}.tile.osm.org/{z}/{x}/{y}.png', {
      attribution: 'OSM'
    }).addTo(map);

    var myMarker = L.marker([startlat, startlon], {
      title: "Coordinates",
      alt: "Coordinates",
      draggable: true
    }).addTo(map).on('dragend', function() {
      var lat = myMarker.getLatLng().lat.toFixed(8);
      var lon = myMarker.getLatLng().lng.toFixed(8);
      var czoom = map.getZoom();
      if (czoom < 18) {
        nzoom = czoom + 2;
      }
      if (nzoom > 18) {
        nzoom = 18;
      }
      if (czoom != 18) {
        map.setView([lat, lon], nzoom);
      } else {
        map.setView([lat, lon]);
      }
      document.getElementById('lat').value = lat;
      document.getElementById('lon').value = lon;
      myMarker.bindPopup("Lat " + lat + "<br />Lon " + lon).openPopup();
    });

    function chooseAddr(lat1, lng1) {
      myMarker.closePopup();
      map.setView([lat1, lng1], 12);
      myMarker.setLatLng([lat1, lng1]);
      lat = lat1.toFixed(8);
      lon = lng1.toFixed(8);
      document.getElementById('lat').value = lat;
      document.getElementById('lon').value = lon;
      myMarker.bindPopup("Lat " + lat + "<br />Lon " + lon).openPopup();
    }

    function myFunction(arr) {
      var out = "<br />";
      var i;

      if (arr.length > 0) {
        for (i = 0; i < arr.length; i++) {
          out += "<div class='address' title='Show Location and Coordinates' onclick='chooseAddr(" + arr[i].lat + ", " + arr[i].lon + ");return false;'>" + arr[i].display_name + "</div>";
        }
        document.getElementById('results').innerHTML = out;
      } else {
        document.getElementById('results').innerHTML = "Sorry, no results...";
      }
    }

    function addr_search() {
      var inp = document.getElementById("addr");
      var xmlhttp = new XMLHttpRequest();
      var url = "https://nominatim.openstreetmap.org/search?format=json&limit=3&q=" + inp.value;
      xmlhttp.onreadystatechange = function() {
        if (this.readyState == 4 && this.status == 200) {
          var myArr = JSON.parse(this.responseText);
          myFunction(myArr);
        }
      };
      xmlhttp.open("GET", url, true);
      xmlhttp.send();
    }
    </script>
  </body>
</html>
``` 

</p>
</details>