tbnobody/OpenDTU #1204: Missing Model HMS-1600

codebomber, 2023-08-03:
### What happened?

while opening the inverter info, it tells me, that my model is unknown.
And It tells me to open a new bug report with the Hardware Number

My Inverter is the Hoymiles HMS-1600-4T
Hardware ID 124097

### To Reproduce Bug

opening inverter Info

### Expected Behavior

It tells me the exact information of my inverter 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.7.10

tbnobody, 2023-08-11:
First of all, please upgrade to the latest version and doublecheck if the error still exists... if so, please provide a screenshot of the whole inverter info dialogue.

tbnobody, 2023-08-25:
Closed due to inactivity

closed 2023-08-25 by tbnobody