tbnobody/OpenDTU #1209: MQTT does not connect to fqdn target

the2Faces, 2023-08-04:
### What happened?

Hi all,

tried to use fqdn information, to force the connection from OpenDTU to the Home Assistant device for MQTT.
But it seems like, that OpenDTU is not able to resolve the hostnames/fqdn entries.
The name of the device was equal to hassio01.domain.local.
The according fqdn was setup on the OPNSense within the local DNS. So the OpenDTU should be able to resolve the right hostname/IP for the mentioned fqdn.

### To Reproduce Bug

Setup a fqdn for the MQTT integration in the setup screen configuration/MQTT and as value for "hostname".
In my example, it was hassio01.domain.local.

### Expected Behavior

With the fqdn setup, OpenDTU should resolve the correct IP for the home assistant.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

f3e3ec0

tbnobody, 2023-08-05:
`.local` seems to be a mDNS target. This has nothing todo with a normal DNS. Only DNS is supported!

LennartF22, 2023-08-05:
I can reproduce the issue. My local domain name is not `.local`, but for testing purposes I created two static DNS entries on my local DNS server: `hassio01.domain.local` and `hassio01.domain.local2`. Both are resolved correctly by `nslookup`, but in OpenDTU, MQTT only seems to be able to connect to `hassio01.domain.local2`.

LennartF22, 2023-08-05:
Probably caused by the `CONFIG_LWIP_DNS_SUPPORT_MDNS_QUERIES` build flag of the default build of the Arduino ESP32 framework.

tbnobody, 2023-08-05:
Seems so. `.local` is somehow pre-defined for mDNS (https://en.wikipedia.org/wiki/.local)
So most likley not a problem with OpenDTU.

closed 2023-08-05 by tbnobody

the2Faces, 2023-08-10:
Hi all,

thanks for the look into this.
I was never experiencing in my local networks any bigger issues, with the dns resolving by using the fqdn domain.local.
But after reading this here it could make sense.
Thanks also for the Wiki entry, I always knew, that Microsoft tends to negate their own assumptions, just in regards of the .local tld exention.
I will try to change my network from .local to .lan and will test again.

Maybe, it is worth commenting within the opendtu dns settings section, not to use the *.local tld extention.