tbnobody/OpenDTU #1221: "nRF24L01+" is shown as "nRF24L01"

CommanderRedYT, 2023-08-07:
### What happened?

My "nRF24L01+" sometimes gets discovered as a "nRF24L01". But the transmission still works and i get all the values from my Inverter. Also, the message "Could not connect to a configured radio module. Please check the wiring." appears while the values are getting updated.

### To Reproduce Bug

Plug a esp32 with a nRF24L01+ into some USB Power Source. Sometimes it shows up as nRF24L01 and sometimes as nRF24L01+

### Expected Behavior

Always show up as nRF24L01+

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.8.2

### Anything else?

I noticed that when for example I reboot OpenDTU through the Webinterface, if it was "nRF24L01+" before, its "nRF24L01" after it has started again. If i unplug it from power and plug it in again it shows "nRF24L01+" again.

tbnobody, 2023-08-07:
This sounds like a hardware issue and not a software issue. Sounds like bad wiring or poor power supply. This is tested with multiple ESPs and NRF modules and never produced any wrong output. The type is determined of the hardware register itself. In this case it would mean that the hardware returns wrong values (for whatever reason).

tbnobody, 2023-08-21:
I am closing this due to inactivity and due to the fact that it's most likely a hardware issue.

closed 2023-08-21 by tbnobody

CommanderRedYT, 2023-08-24:
@tbnobody I tested it a bit more, and it seems to only occur when restarting without power cycling. So, if OpenDTU crashes or performs a reboot, it will be the non-plus version, but if i perform a power-cycle, it shows the plus version. Therefore I do not believe it is hardware related.