tbnobody/OpenDTU #1234: missing model HMS-2000

wabildi, 2023-08-11:
### What happened?

all works fine, but not all information about the inverter are shown under inverter information.

inverter part numer : 124097

Only see:
serial number
production year
etc

not seen:

how much watts peek 
modell information

### To Reproduce Bug

no inverter information if you click on the inverter invormation

### Expected Behavior

please impliment the fll HMS-2000 support, or tell me how to impliment it by myself.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.6.1

### Anything else?

big thanks to this community.

tbnobody, 2023-08-11:
First of all, please upgrade to the current version and doublecheck if the problem still exists.

tbnobody, 2023-08-25:
Closed due to inactivity

closed 2023-08-25 by tbnobody