tbnobody/OpenDTU #1235: MQTT - Übertragung Summenfelder "Gesamtertrag Insgesamt", "Gesamtertrag Heute", "Gesamtleistung" [Request]

wesseli01, 2023-08-12:
### Is your feature request related to a problem? Please describe.

Leider werden nicht die Summenfelder  "Gesamtertrag Insgesamt", "Gesamtertrag Heute", "Gesamtleistung" (wie auf der Live-Übersicht sichtbar) jeweils per MQTT übertragen. Wenn man mehrere DTU's aufgenommen hat, dann muss man die Einzelwerte erst aus der Einzel-DTU in der Drittsoftware (z.B. Home Assistant) addieren und in einer neu angelegten Variable ablegen. 

### Describe the solution you'd like

Aufnahme der bereits vorhandenen Summenfelder aus der Live-Übersicht jeweils als Variable im MQTT

### Describe alternatives you've considered

Addition der betreffenden Einzelvariablen aus dem MQTT-Protokoll der jeweiligen DTU in der Drittsoftware

### Additional context

[image]

wesseli01, 2023-08-12:
@LennartF22, Hättest Du da auch eine Idee, da Du uns ja so kompetent bei der MQTT-Namens #1194 geholfen hast?

Kampfschildkroete, 2023-08-13:
Vielleicht verstehe ich es falsch, aber das müsste die topics _ac/yieldtotal_ und _ac/yieldday_ sein.

Siehe hier: https://github.com/tbnobody/OpenDTU/blob/master/docs/MQTT_Topics.md#inverter-total-topics

wesseli01, 2023-08-13:
Diese Felder werden leider nur für die Anzeige der jeweiligen Yieldtotal und Yieldday-Werte der angeschlossenen Einzel-WR genutzt. Das sind nicht o.g. Summenfelder in der Live-Übersicht, die die Einzelwerte der überwachten WR korrekt addieren. Diese fehlen im MQTT-Protokoll.
[image]

Es werden diese Summenfelder benötigt.

[image]

turrican944, 2023-08-14:
Moin
Die Summen werden auch übertragen, die musst du dir nur selbst im Homeassistant einbinden, das Autodiscovery legt die Sensoren (zurzeit) nicht an. Musst du über die YAML die entsprechende MQTT Sensoren anlegen.
Ist in der MQTT.,md auch beschrieben

Topic  
solar/ac/power   Sum of AC active power of all enabled inverters 
solar/ac/yieldtotal    Sum of energy converted to AC since reset watt hours of all enabled inverters 
solar/ac/yieldday     Sum of energy converted to AC per day in watt hours of all enabled inverters

wesseli01, 2023-08-16:
@turrican944 
Leider kann ich keine MQTT.,md finden. Lediglich die Datei [https://github.com/tbnobody/OpenDTU/blob/master/docs/MQTT_Topics.md](url) finde ich. Dort kann ich aber auch keine Beschreibung zu der Anlage und der Übertragung der Felder finden. Aktuell werden die Felder lediglich für die Einzelpanels genutzt. Kannst Du mir den Link und die Beschreibung senden? Danke.

turrican944, 2023-08-16:
Ja diese Datei meinte ich, du musst die topic s der gesamt Leistung usw (das was du haben möchtest) selbst im Home-Assistent einbinden, in deiner configuration.yaml oder mqtt.yaml.

wesseli01, 2023-08-16:
Das hatte ich verstanden. Aber wo ist die von Dir genannte Beschreibung dazu? Denn diese Felder werden doch aktuell als Einzelwerte für die Panels verwendet.

closed 2023-08-16 by wesseli01

KlausHans, 2024-01-29:
Ich bin auch über dieses Thema gestolpert und war schon kurz davor die Werte in HA manuell zu addieren. Zum Glück noch mal gegoogelt.

> das Autodiscovery legt die Sensoren (zurzeit) nicht an. 

Könnte das in Zukunft implementiert werden? Das wäre schön, weil in der Config-yaml arbeite ich nur ungern und es ist ein extra manueller Schritt.

turrican944, 2024-01-29:
Moin
Musst ja nicht unbedingt die config.yaml bearbeiten, du kannst dir auch eine Automation anlegen die beim starten vom Home Assistent die passende discovery Message in den Broker schickt , dann legt Homeassistent die Sensoren.
Ich finde das Beispiel nur Grade nicht und ich mache das bei mir über die yaml.

KlausHans, 2024-01-29:
Auch ne Idee, hab ich aber noch nie gemacht. Habs jetzt in die config gepackt weil ich so ähnlich schon andere Entitäten hab. Aber will ich eigentlich lieber vermeiden ;)