tbnobody/OpenDTU #1237: In HomeAssistant it is not possible to add the Hoymiles Temperatur as Statisticsdiagram

pioneer-01, 2023-08-13:
### What happened?

When configuring in HomeAssistant a Statistics diagram, it s not possible to add the temperature from Hoymiles HM600 coming via MQTT. It is not shown in the list at all. 
According to HomeAssistant it is a bug of the entity. 
see screenshot and URL. 

### To Reproduce Bug

Create Statistics Diagram. 
add an entity —> Hoymiles temperature is missing in the list

### Expected Behavior

The temperature can get selected. 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

23.8.8

pioneer-01, 2023-08-13:
See attached screenshot
https://ibb.co/n7B4kdy

knurrdog, 2023-08-13:
I've no problem with Home Assistant Version 2023.8.2 adding a statistics diagram with HM temperature.
What version of HA do you use?
[Screenshot](https://ibb.co/5xt19Yp)

pioneer-01, 2023-08-13:
Sorry guys! Was my fault. I tried to combine Power statistics with temperature, which is not possible. If I select temperature it works fine.

closed 2023-08-13 by pioneer-01