tbnobody/OpenDTU #1238: Reaction once WLAN stops working

KemitFrog, 2023-08-14:
### Is your feature request related to a problem? Please describe.

Hello,
I encountered yesterday that once WLAN stops working, my inverter is noch contolable anymore. I find this somehow safety critical as I use the DTU for power limitation. As my automation software ioBroker will not be able communicate anymore with the DTU, I wouldn't know where else to solve this than OpenDTU

### Describe the solution you'd like

It would be greate if after a timeout, the inverter would be shut off or limit the power to a configurable value.

### Describe alternatives you've considered

Direct wireless connection from the automation software to the DTU not needing the router.

stefan123t, 2023-08-16:
Granted a safety valve to limit inverters to a certain power output would be nice to have, especially for oversized models according to the local law on export limits (600/800W in most EU countries, but zero export in some others).

Did you consider setting the persistent limit to that low / zero value and only set a non-persistent limit via ioBroker / OpenDTU ?
It may be an option to use a more stable wired connection either using an Olimex or openDTU Fusion Shield with PoE Ethernet.

Also checkout the spin-off / fork from Helge Erbe which is called [OpenDTU-onBattery](https://github.com/helgeerbe/OpenDTU-OnBattery/) maybe he has something implemented to that ends already ?

stefan123t, 2024-08-05:
@KemitFrog please double check the **Dynamic Power Limiter (DPL)** function in the daughter project[ OpenDTU-OnBattery](https://github.com/helgeerbe/OpenDTU-OnBattery/).

I have described a [possible PM+DPL setup](https://github.com/tbnobody/OpenDTU/issues/272#issuecomment-2219644352) with a Shelly Pro3EM as **Power Meter (PM)** here in another issue.