tbnobody/OpenDTU #1249: webapi: It is not possible to switch off the inverter

rainer2830, 2023-08-16:
### What happened?

Environment:
2 inverters (HM800, HM1500)
1 DTU
MQTT enabled
HomeAssistant disabled
password set

MQTT: Switching the inverters off and on works.
webAPI: Only switching ON works via the webAPI, but not switching OFF.

### To Reproduce Bug

Step 1:
Open the terminal, enter the following command (adjust username, password, ip and serial) and confirm:
curl -u username:password http://192.168.178.2/api/power/config -d 'data={"serial":"1234567890", "power":"0"}'

Return message: {"type":"success","message":"Settings saved!","code":1001}

Step 2:
Check status in openDTU webgui (incl. waiting time)
Result: Inverter is on and nothing happened

Step 3:
Switch off the inverter via openDTU webgui and wait until the inverter is off.

Step 4:
Open the terminal, enter the following command (adjust username, password, ip and serial) and confirm:
curl -u username:password http://192.168.178.2/api/power/config -d 'data={"serial":"1234567890", "power":"1"}'

Return message: {"type":"success","message":"Settings saved!","code":1001}

Step 5:
Check status in openDTU webgui (incl. waiting time)
Result: Inverter is on!

### Expected Behavior

in short: the inverter should also be able to be switched  off via webapi

Expected Behavior:

curl -u username:password http://192.168.178.2/api/power/config -d 'data={"serial":"1234567890", "power":"0"}'

--> result: Inverter is power off

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.8.8

tbnobody, 2023-08-16:
Is it working if you just USE the web interface instead of the console? If it's working here (and for me it does) then your curl command might be wrong.

rainer2830, 2023-08-17:
webinterface and MQTT is working.
OK, following your hint I tried a few data types:

int: 
..."power":0}' --> OK
..."power":1}' --> OK

string 
..."power":"0"}' --> nicht OK
..."power":"1"}' --> OK

bool 
... "power":false}' --> OK
... "power":true}' --> OK

bool as string 
..."power":"false"}' --> nicht OK
..."power":"true"}' --> OK

So it works in principle and doesn't seem to be a general bug...

closed 2023-08-17 by tbnobody