tbnobody/OpenDTU #1253: Wenn man Limit 600 W setzt, kommen nur 50 % bzw. 300 W

tvellen, 2023-08-20:
### What happened?

Man kann das Limit setzen und es wirkt auch in wenigen Sekunden, aber es reduziert sich auf 50% vom gesetzten Limit.

### To Reproduce Bug

OpenDtuFusionV2 FusionBoard
HMS 2000
nur 2 Module mit 550 Watt, keine möglichen 4 Module
volle Sonne, also mehr als 600 Watt

### Expected Behavior

Wenn man das Limit auf 600 W setzt kommen auch 600 W statt 300.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.8.8 und v23.7.12

DPO99, 2023-08-20:
Das klingt danach als ob der das Limit trotz 4 tracker auch auf jew. 2 Eingänge drosselt.  Sind beide Module auf einer Seite angeschlossen? Ich kenne den HMS nicht aber beim HM wird das so gemacht. Somit kein Fehler der DTU sondern ein Verhalten vom Hoymiles.

tvellen, 2023-08-20:
Der HMS 2000 hat 4 Tracker. Die zwei Module sind also kein Pärchen, auch wenn sie an einer von zwei Seiten hängen.

Habe 3. Modul angeschlossen und 600 W eingestellt. Dann kamen nicht 300 W, sondern 450 W, obwohl ohne Beschränkung über 1000 W verfügbar waren.

D.h. wohl, dass nicht eine Summe pro Wechselrichter beschränkt wird, sondern prozentual pro Modul.

Das bedeutet, die Funktion von openDTU suggeriert etwas Anderes als tatsächlich passiert. Es wäre also besser, wenn openDTU es anders beschriften würde, z.B.  nur % pro Modul.

Wenn das bei verschiedenen Wechselrichtern unterschiedlich ist, dann ist das halt nicht einheitlich und müsste auch von openDTU unterschieden werden.

Wenn das richtig ist, dann weiß ich jetzt aber wie ich die richtige Beschränkung hinkriege.

(Ist halt blöd von Hoymiles, wenn ich zwei Module nur auf 300 Watt beschränken kann, wenn die Module unterschiedliche Ausrichtungen haben.)

DPO99, 2023-08-20:
Das ist pauschal nicht möglich. Der HM1500 z.b. hat nur 2 tracker und stellt dann die Leistung pro tracker also für 2 Module ein. Ansonsten müsste man diese Anzeige Modellabhängig machen aber das ändert ja nichts am Problem. Ggf. Reicht ja schon eine Info im Wiki, falls es da nicht schon steht.

tvellen, 2023-08-20:
Hmm, kenn mich mit den (anderen) Wechselrichtern nicht aus. Haben die verschiedenen eine dokumentierte API? Oder sind das immer "re-engineerte" Codes ohne Namen/Beschreibung? Oder was sagt die API? Jedenfalls, suggeriert openDTU, ich könnte 600 Watt als Summe setzen, funktioniert aber nicht. Stattdessen kann ich offenbar % pro Tracker setzen. Kann man natürlich in der Doku beschreiben oder besser in der App klarer machen.

olivierschmid, 2023-09-15:
Ja - das ist bei mir auch so. Ich betreibe 3 Panels an einem HM1500. Somit ist ein Input nicht benutzt. Wenn ich auf 600W limitieren will muss ich somit 800W eintragen (3 x 200W + der leere Input)

nakott, 2023-10-11:
Hallo,
bei Homyles WR wird das gesetzte Limit durch die Anzahl an Tracker geteilt. Das ist dann das Limit je Tracker. 
Warum wird das so gemacht? Ganz einfach, denn der Tracker kann ganz einfach das Modul in den Spannungsbereich steuern, damit das Modul genau die gewünschte Leistung liefert. Beispiel. Umpp läge bei z.B. 35V bei einem 380 Wp Modul und das Limit für einen Tracker läge bei 200 W. Dann wird der Tracker die Spannung jenseits von Umpp fahren, z.B. bei 38-40V. Dazu einfach die Kennlinien der Module schauen.
Beim HM-1500 mit Tracker und max 750 W je Tracker hängt man ein West und ein Ost Modul an einen Tracker. Voraussetzung: Es sind zwei identische PV-Module. Denn bei PV Modulen ist Umpp bei geringer und hoher Einstrahlung fast gleich. Bei Drosselung auf 600 W hat jeder Tracker ein Limit von 300W und so hat man von morgens bis abends durchgehend 600 W.
Das Interface zwischen OpenDTU und Homyles liegt das Limit immer bezogen auf den ganzen WR fest. Und der WR teilt dann intern durch die Anzahl an Tracker. Wenn man das weiß, dann ist die Anzeige in OpenDTU gut so.

nakott, 2024-02-06:
Danke für den Link zum Video. Vielleicht nochmal einen Schritt zurück. Aus Sicht der OpenDTU ist heute alles richtig. Sie setzt mit einem API Call das Limit im WR. Damit ist für die OpenDTU der Fall erledigt. Mehr Möglichkeiten gibt es für die OpenDTU nciht. Homyles ordnet dann intern ein festes Limit jedem Tracker zu; und zwar Gesamtlimit / Tracker. Das Video zeigt nun, dass die 4. Generation von Deye eine dynamische Verteilung auf die Tracker vornimmt. Das ist eine positive Weiterentwicklung und vielleicht zieht Homyles in der Zukunft damit auch nach. Das wäre sicher zu begrüßen. Aber das sind keine Dinge, die man der OpenDTU "anlasten" kann. Das wäre eine Diskussion mit Homyles. Man bedanke aber auch: Mit Homyles und der OpenDTU kann sich wunderbar ein lokales Monitoring ganz ohne Cloud in China aufbauen, die Daten mit anderen Daten aus dem Haus kombinieren etc. Für mich ist das der entscheidende Vorteil. Es ist aber gut, dass die Info zu Deye jetzt so verfügbar sind und so kann jeder den für sich geeigneten WR nutzen.

nakott, 2024-02-06:
Mit 3 Module am HMS2000 (4 Module - 4 Tracker) wäre ein Limit von 800 vermutlich sinnvoll, denn real wären das dann 3 * 200 = 600 W. Ob man eine solche Logik in die OpenDTU integrieren sollte sehe ich eher problematisch. Denn wenn Homyles zur 4. Generation von Deye via FW-Update gleichziehen könnte, dann müsste die OpenDTU für eine Umrechnung auch noch die FW Version berücksichtigen. Wenn man die Thematik kennt, kann man heute das Limit geeignet setzen. Aber klar, wem dieses Thema nicht bewusst ist, der kann sich schon sehr wundern. Und das neue Verfahren von Deye ist intuitiver.

tbnobody, 2024-03-31:
it's not a bug, it's a feature bzw. firmware limitation

closed 2024-03-31 by tbnobody