tbnobody/OpenDTU #1258: YieldTotal calculation invalid at end of day or when inverter restarts.

broth-itk, 2023-08-21:
### What happened?

The YieldTotal value calculation is returning bad results when the inverter restarts, usually at end of day.

Synopsis: When I go to OpenDTU Live View in the evening, I often see only very low YieldTotal values.

Looking to the inverters, the 4-channel unit has often restarted due to low voltage. This causes the "last received value" of YieldTotal to be zero.

Not only the live dashboard is affected but the wrong values are getting published to MQTT as well.
Grafana will show disappointing values :)

### To Reproduce Bug

Just restart your inverter and observe the values.
YieldTotal will decrement.

### Expected Behavior

If inverter resurrects after shutdown for whatever reason, check if the last known value of YieldTotal is greater than the "new" value.
If smaller, do not update statistics.

At midnight, all daily stats values should be reset anyway.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.8.8

broth-itk, 2023-08-23:
Proposed fix with https://github.com/tbnobody/OpenDTU/pull/1267

OneCreek, 2023-08-31:
I think I have the same/similar issue...

thats the values I get via MQTT:
[image]
[image]
[image]
[image]
[image]

broth-itk, 2023-08-31:
The fix currently only addresses the AC YieldTotal value for each inverter.

If the proposed fix has been checked and validated by others, it can be extended to work the same way for each individual inputs (channel/string/module) as well.

commit 1de3b481 (2023-12-16): Feature: Implement offset cache for "YieldDay"

Thanks to @broth-itk for the idea!
Fix: #1258 #1397
closed 2023-12-16 by tbnobody