tbnobody/OpenDTU #1275: [Request] Support LORA to overcome WIFI range issues

fakegit997, 2023-08-26:
### Is your feature request related to a problem? Please describe.

My inverter is quite far away so I am not able to bridge the distance to my router with WIFI and NRF24L01+. 

### Describe the solution you'd like

If opendtu would support LORA, range issues could be easily resolved. LORA MCUs are cheap and powerful with integrated wifi. So all features of opendtu could be supported + upload to lorawan.

### Describe alternatives you've considered

Expanding my Wifi to the roof is overkill to me.

Kampfschildkroete, 2023-08-27:
I know that isn't an answer to your request but do you have an ethernet connection available in the roof? If yes you could use an ESP32 that supports an ethernet connection.

tbnobody, 2023-08-27:
Lora requires an additional RF hardware, different protocol implementation, and a lot of data to be transfered (not sure how this copes with the duty cycle)

fakegit997, 2023-08-27:
@Kampfschildkroete : I Agree and it would be an option in my case. Still, I think the request is valid
@tbnobody : 

> Lora requires an additional RF hardware

On the DTU side it doesn't if you use boards with Lora already soldered on. I've just tested a TTGO ESP32 with OpenDTU and it works just fine. Of course there must be a Lora gateway somewhere around, but I think a lot of the users live in a covered area and could easily use TTN to get their data. 

> and a lot of data to be transfered

I agree partially. It heavily depends on the use case. Lora wouldn't be suitable for polling rates of e.g. 5s. But for hourly data transfer it shouldn't be an issue. 
Side effect:  openDTU users are generally techies so they may boost Lora coverage by running their own gateways on TTN

stefan123t, 2023-10-14:
I love LoRaWAN too, but would prefer a openDTU Relay mode more to be honest. 

Also Hoymiles offers DTU repeaters and in order to implement support for the newer HMS-WiFi series we will need to implement the so called NetworkCommand set from the DTU Pro, S-Miles Installer App and the S-Miles Cloud API anyway. Here a repeater mode of the DTU can act as multiple things:

* DTU repeater simply forwarding commands from the main OpenDTU to the inverter and back again (Hoymiles Gen2/Gen3 protocol)
* DTU which can be discovered in S-Miles cloud and which can be used to trace additional features, eg Grid Profile or Firmware Download / Install (Hoymiles NetworkCommand)
* mock Inverter which we can use to test new protocol command implementations against (Hoymiles Gen2/Gen3 protocol)
* mock DTU which we can use to test the not yet implemented S-Miles Cloud API against (Hoymiles NetworkCommand)
* mock DTU-BI (built in) which is integrated into the new Hoymiles HMS-WiFi models (Hoymiles NetworkCommand)

Some of these could be tested with the S-Miles Installer app or the S-Miles Cloud.

stefan123t, 2024-08-05:
@tbnobody maybe we should start to implement a serial protocol, which we can use both via UART/USB and / or via the Info > Console in order to send dedicated commands to the DTU and the inverter and that returns the HEX or parsed data from the inverter in verbatim ?

This could then be extended to provide a core for OpenDTU which can be called via REST API, MQTT, UART/USB Console as well as unattended as before by consecutively querying all the inverters as setup via the UI and queried by the livedata page.

stefan123t, 2025-06-27:
@c-e-github I have also started to build a OpenDTU fork with LoRaWAN / the things network support.

I managed to get OpenDTU build for my Heltec ESP32 LoRa 32 V2 board, i.e. using

<details><summary>Details platformio.ini</summary>
<p>

```ini
lib_deps= ...
     mcci-catena/MCCI LoRaWAN LMIC library@^5.0.1
...
[env:heltec_wifi_lora_32]
board = heltec_wifi_lora_32
build_flags = ${env.build_flags}
    -DPIN_MAPPING_REQUIRED=1
    # requirements for MCCI LoRaWAN LMIC library
    -D ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS
    -D CFG_eu868=1
    -D CFG_sx1276_radio=1
    -D DO_WORK_INTERVAL_SECONDS=60
    ; -D ABP_ACTIVATION                ; Use ABP instead of OTAA activation.
    ;
    ; -D WAITFOR_SERIAL_SECONDS=10     ; Can be used to override the default value (10).
    ;                                    Is used only for boards with default set to != 0 in BSF.
    ;
    ; -D LMIC_CLOCK_ERROR_PPM=0        ; If not defined defines, otherwise overrides value defined in BSF.
    ;                                    Is for testing purposes only.
    ;                                    Do not enable this unless you explicitly know what you are doing.
    ;
    ; -D STM32_POST_INITSERIAL_DELAY_MS=1500  ; Workaround for STM32 boards. Can be used
    ;                                           to override value (milliseconds) in BSF.
    -D hal_init=LMICHAL_init           ; Workaround for naming conflict of function hal_init
                                       ; introduced by newer versions (> 3.5.0) of
                                       ; PlatformIO Espressif 32 platform (ESP32 Arduino core).
                                       ; See https://github.com/lnlp/LMIC-node/issues/41 for more information.
    -DSCALE_ENABLED=false
    -DSCALE_DATA_PIN=2
    -DSCALE_SCK_PIN=3
    -DSCALE_CALIBRATION_FACTOR=1.0f
    -DSCALE_OFFSET=0L
```

</p>
</details> 

I have added the following Pin Mapping for the Board and the connected SSD1306 Display.

<details><summary>Details pin_mapping.json</summary>
<p>

```json
[
    {
        "name": "Heltec WiFi LoRa ESP32 v1/v2 with SSD1306 Display",
        "links": [
            {"name": "Information", "url": "https://docs.platformio.org/en/latest/boards/espressif32/heltec_wifi_lora_32_V2.html"}
        ],
        "nrf24": {
            "miso": -1,
            "mosi": -1,
            "clk": -1,
            "irq": -1,
            "en": -1,
            "cs": -1
        },
        "cmt": {
            "clk": -1,
            "cs": -1,
            "fcs": -1,
            "sdio": -1,
            "gpio2": -1,
            "gpio3": -1
        },
        "eth": {
            "enabled": false,
            "phy_addr": -1,
            "power": -1,
            "mdc": -1,
            "mdio": -1,
            "type": 0,
            "clk_mode": 0
        },
        "w5500": {
            "mosi": -1,
            "miso": -1,
            "sclk": -1,
            "cs": -1,
            "int": -1,
            "rst": -1
        },
        "sx1276": {
            "nss": 6,
            "rxtx": -1,
            "rst": 5,
            "dio0": 2,
            "dio1": 3,
            "dio2": 4
        },
        "hx711": {
            "data": -1,
            "sck": -1
        },
        "led": {
            "led0": 25,
            "led1": -1
        },
        "display": {
            "type": 2,
            "data": 4,
            "clk": 15,
            "reset": 16
        }
    },
...
```

</p>
</details>