tbnobody/OpenDTU #1277: [Request] Ability to set cpu frequency and set cpu frequency to lowest stable value by default

ss89, 2023-08-27:
### Is your feature request related to a problem? Please describe.

i would like my smart devices to consume as less energy as really needed.

when looking at the status page of my dtu, i saw that it clocks at 240mhz which i found quite high for just handling a bit of network traffic each few seconds and rest of the time sleeping

### Describe the solution you'd like

I would like to have opendtu set the lowest safe value for cpu clock as possible and an option in the settings to increase it, should this default value not work for someone

### Describe alternatives you've considered

waste energy 🙈

tbnobody, 2023-08-27:
And you are of course also aware of the fact that the frequency of the NRF module has to be changed every 4ms to receive the data from the inverter? You can try it to change the software for your needs. It's open source, just do it...

ss89, 2023-08-27:
I am not aware of that, because i didn't analyze the software for the need.

Since you are providing a feature request template for this repo, i thought i'd give it a try.

closed 2023-08-27 by tbnobody