tbnobody/OpenDTU #1278: Crash and reboot during boot

T3rm1, 2023-08-27:
### What happened?

My ESP32-WROOM-32E no longer boots with newest [commit](https://github.com/tbnobody/OpenDTU/commit/08ca2214102ea0c3fd9cd5fbc496d391f0e7ed3f).

Looks like it crashes when trying to open the configuration file.

<details>
  <summary>Log for uploading image</summary>

```
Processing generic (board: esp32dev; framework: arduino; platform: espressif32@6.3.2)
--------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Firmware Revision: v23.8.22
No custom_patches specified
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.3.2) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 47 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP Async WebServer @ 1.2.3+sha.1dde9cf
|-- ArduinoJson @ 6.21.3
|-- espMqttClient @ 1.4.4+sha.f633972
|-- RF24 @ 1.4.7
|-- U8g2 @ 2.35.4
|-- sunset @ 1.1.7
|-- LittleFS @ 2.0.0
|-- Hoymiles @ 0.0.1
|-- Ethernet @ 2.0.0
|-- Update @ 2.0.0
|-- ResetReason
|-- TimeoutHelper
|-- DNSServer @ 2.0.0
|-- WiFi @ 2.0.0
|-- MqttSubscribeParser
|-- Ticker @ 2.0.0
Building in release mode
Retrieving maximum program size .pio\build\generic\firmware.elf
Checking size .pio\build\generic\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  20.0% (used 65576 bytes from 327680 bytes)
Flash: [========  ]  77.9% (used 1531013 bytes from 1966080 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: COM3
Uploading .pio\build\generic\firmware.bin
esptool.py v4.5.1
Serial port COM3
Connecting....
Chip is ESP32-D0WD (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 8c:ce:4e:89:64:b0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
[... 143 lines cut ...]
12:39:57.663 >
12:39:57.663 >
12:39:57.663 >
12:39:57.663 >
12:39:57.663 > ELF file SHA256: 9b90468d1745b5b0
12:39:57.665 >
12:39:57.665 > E (472) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
12:39:57.665 > Rebooting...
```

### To Reproduce Bug

Just compile code and upload to ESP32.

### Expected Behavior

No boot loop.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

08ca2214102ea0c3fd9cd5fbc496d391f0e7ed3f

tbnobody, 2023-08-27:
I've just used a new ESP (as you did because you don't have any config data on your partition) and it worked without any problems. 
Have you tried to erase the ESP and completly reprogram it?

T3rm1, 2023-08-27:
Yes, I erased it with the PlatformIO extension. Did that multiple times. Same result.

tbnobody, 2023-08-27:
I don't think that the erase was successfull... Otherwise you should see a 
```
Starting OpenDTU
Initialize FS... ./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
E (18) esp_littlefs: mounfft failed,  (-84)
```
at the first boot. Did you also try a power cycle?

Not sure what's wrong with your ESP, but it doesn't seem to be a general issue.

T3rm1, 2023-08-27:
I erased it several times and also uploaded it several times. Maybe I didn't do it when I logged the output.
Well, my ESP works with the old version from March. I switched back to that one for now. But since it is working, this usually means that the code changes introduced a bug.

tbnobody, 2023-08-27:
You could try several other versions to determine until which version it works.

T3rm1, 2023-08-27:
Ok, I spent way more time on this than planned but I hope this helps you. Ran a `git bisect` and tested several builds:

```
F:\src\OpenDTU> git bisect good
5bbc67bc49d2fb32298916f23e8c2e5be17078d6 is the first bad commit
commit 5bbc67bc49d2fb32298916f23e8c2e5be17078d6
Author: Thomas Basler <thomas@familie-basler.net>
Date:   Mon Jul 31 22:52:38 2023 +0200

    Fix: Clear parser buffers to prevent random numbers if no data was received

 lib/Hoymiles/src/parser/AlarmLogParser.cpp         | 1 +
 lib/Hoymiles/src/parser/DevInfoParser.cpp          | 2 ++
 lib/Hoymiles/src/parser/StatisticsParser.cpp       | 1 +
 lib/Hoymiles/src/parser/SystemConfigParaParser.cpp | 1 +
 4 files changed, 5 insertions(+)
```
Everything before that commit worked.

stefan123t, 2024-08-06:
@T3rm1 given this issue is open for more than a year, could you try it with one of the latest release versions ?
https://github.com/tbnobody/OpenDTU/releases 
Either 24.08.05 or 24.06.29 should be running fairly fine IMHO, though 24.08.01 should be avoided for some issue introduced with it and fixed yesterday.

closed 2025-02-07 by tbnobody

T3rm1, 2025-02-10:
@stefan123t It still doesn't work with the newest commit as of today. I don't get a crash during boot but when I try to open the web ui it is just a blank page. Maybe the device is faulty.