tbnobody/OpenDTU #1296: MQTT host name DNS lookup ignores TTL setting

sauvant, 2023-09-01:
### What happened?

Had a DNS entry (TTL: 1h) pointing to the IP address of my MQTT broker. OpenDTU client configured to use that host was sending data to that broker. Now I wanted to move the broker to another server. Changed the DNS entry of the corresponding hostname to point to the new servers' IP address. OpenDTU clients that I started after 1h of waiting for the DNS propagation sended the data to the new server as expected. But the first one did never switch to the new broker and continued to use the old IP address and thus the old broker for more than 24h. Only way to convince it was a reset of OpenDTU.

Not sure wether this is OpenDTU code specific or caused by lower level libraries.

### To Reproduce Bug

- Start two MQTT brokers on two hosts:
broker1: IP address 1
broker2: IP address 2

- Setup a DNS entry for host my.example.com to point to IP address 1, TTL: 60 s
- Start an OpenDTU client, configure my.example.com as MQTT hostname.
- See data coming in on broker1
- Now change the DNS entry to point to IP address 2

-> The OpenDTU client will continue to send data to broker1 until forever. Or until is gets resetted manually.

### Expected Behavior

The OpenDTU client is expected to send its data to broker2 after 60 s of configured DNS entry "time to live".

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

 v23.8.8/b71106c

tbnobody, 2023-09-01:
DNS resolution only takes place at connection time. If you never disconnect the client from the broker the client will never change 

> Not sure wether this is OpenDTU code specific or caused by lower level libraries.

In this case I would say it's not an issue but it's absolutley working as expected. Thats how all DNS based connections are working.

sauvant, 2023-09-01:
Hmm. May be true but imagine loads of DTUs out there reporting to one server (central service to supply historical data and dashboards) and that server needs to be moved to another IP. Would require to reboot all clients manually. How about reconnecting periodically, for instance once a day?

sauvant, 2023-09-01:
Or wait, you mean forcing a disconnect by taking down the server would be sufficient to enforce a new name resolution? Did not check that 🤔

tbnobody, 2023-09-01:
> you mean forcing a disconnect by taking down the server would be sufficient to enforce a new name resolution?

Of course. That's how it works. Is independent of who performs the disconnection. If the TTL has expired and a connection takes place a new DNS resolution is performed. (Keep in mind that all DNS caches are also looking at the TTL)

sauvant, 2023-09-01:
Ok, thanks for the clarification. And for your great work here 👍👍👍 I close as completed, no bug.

closed 2023-09-01 by sauvant