tbnobody/OpenDTU #1298: WR "Ausschalten" nach einstellbarer Zeit [Request]

fdmarder, 2023-09-01:
### Is your feature request related to a problem? Please describe.

Problem: 
HomeAutomatition steuert über Ein/Ausschalten des Hoymiles Wechselrichter die Entladung der Batterie.
HomeAutomation stürzte ab und Hoymiles Wechselrichter entlädt Batterie bis auf die Leerlaufspannung.

### Describe the solution you'd like

Einstellbare Zeit  in min, nach der die openDTU den Hoymiles Wechselrichter "Ausschaltet".  Zeit starte ab "Einschalten" .  Zeit = 0 -> kein Timeout wie bisher. 

### Describe alternatives you've considered

Alternative: Wird von der HausAutomation nicht alle x Minuten (einstellbar) ein "Einschalten" (keepalive) gesendet, schaltet die openDTU den Hoymiles aus.

DPO99, 2023-09-03:
Also soll openDTU die Aufgabe deines (vermutlich nicht vorhandenen) BMS übernehmen? Alternativ soll die DTU (gesteuerte Einheit) die Steuerungseinheit überwachen? Du solltest m.E. eher das Problem des abstürzenden HA lösen. Es gibt kein 100% sicheres System. Deswegen solltest du solche Sachen über Notabschaltungen (BMS) oder ein Backup System regeln. Ich nicht verstehen, wieso so viele erwarten, dass die DTU alle Probleme lösen soll....

fdmarder, 2023-09-03:
BMS vorhanden und hat Batterie abgeschaltet. Hatte nirgends etwas von abgestürzter HW geschrieben . 
DTU soll u. kann nicht alle Probleme lösen, doch als Bindeglied zwischen der HausAtomatisierung und dem Hoymiles würde es die Zuverlässigkeit der Gesamtkette  erhöhen. Btw in #1238 wurde ein ähnlicher Wunsch geäußert.

DPO99, 2023-09-03:
> HomeAutomation stürzte ab und Hoymiles Wechselrichter entlädt Batterie bis auf die Leerlaufspannung.

??? Worum geht's dann? 

Sofern du ein BMS hast ist doch alles tutti. Das schlimmste was passieren kann ist also dass dein Akku leer wird. 

Der Entwickler hat schon mehrfach erwähnt, dass es nicht unendlich Platz für alle Lösungen auf dem esp32 gibt. Für die Allgemeinheit ist das scheinbar nicht interessant (gibt ja auch keine Reaktionen auf den Beitrag).

Darüber hinaus kann man sowas auch über einen Shelly oder dergleichen lösen, der AC seitig trennt, wenn die DTU nicht erreichbar ist. Also nix was m.E. den wenigen Platz auf dem esp32 belegen sollte.

fdmarder, 2023-09-03:
danke für deine hilfreichen Hinweise.  Vielleicht sollte man den Entwickler selbst entscheiden lassen.