tbnobody/OpenDTU #1304: Limit nonpersistent über MQTT überschreibt Limit persistent

sholvatealc, 2023-09-03:
### What happened?

Unter Homeassistant ändere ich die Limits des Hoymiles HM-1500 um eine zu hohe Netzeinspeisung zu vermeiden.
Eine Änderung des Limit "nonpersistent" entweder als absolute oder relativ ändert aber auch das Limit "persistent".
Dies ist auch in den Logs von HA zu sehen.
[image]

Das Limit nach unten setzen geht. Aber nicht mehr nach oben.

### To Reproduce Bug

In HA unter MQTT das Limit nonpersistant ändern sieht man wie sich der persistant mit ändert.
[image]

### Expected Behavior

...............

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

e1c2817 von Lenhard wegen der Namensrichtlinie

tbnobody, 2023-09-03:
> In HA unter MQTT das Limit nonpersistant ändern sieht man wie sich der persistant mit ändert.

Das ist exakt wie es funktionieren soll. Beim Auslesen des Wechselrichters kann nicht zwischen einem permanenten Limit oder einem nicht-permanenten Limit unterschieden werden. Der Wechselrichter gibt einfach nur ein Limit zurück.

sholvatealc, 2023-09-03:
Danke für die Antwort.
Eine Frage noch.
Wenn ich von HA aus den nonperistant Wert ändere landet dieser aber im
flüchtigem Speicher?
Will nicht den Wechselrichter auf Dauer kaputt machen.

Grüße Markus

tbnobody, 2023-09-03:
> Wenn ich von HA aus den nonperistant Wert ändere landet dieser aber im flüchtigem Speicher?

Korrekt

sholvatealc, 2023-09-03:
Danke für die Info. Kann ich das als geschlossen markieren ? Oder Sie ? Weiß nicht wie das geht. .-))

closed 2023-09-03 by tbnobody