tbnobody/OpenDTU #1310: Connection lost Hoymiles hms 2000

Mikespeed2023, 2023-09-04:
### What happened?

Each day everything starts correct. at any time it begins 2 lose the connection.

### To Reproduce Bug

Just waiting. I have a shelly plug between hoymiles. So i see hoymiles is still working.

### Expected Behavior

I realy don´t know. The problem siappers after 30 min. and begin again.
[image]
[image]

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

	v23.8.28

### Anything else?

The sistance between hoymiles and dtu is about 110 cm. iah the problem with different firmwares.

tbnobody, 2023-09-04:
have to tried to change the frequency a little bit? Are you using a proper antenna? What is your configured RF power?

Mikespeed2023, 2023-09-04:
I didn´t try a different antenna beacuse it wprked for 3 weeks very good. O reduced rge distance frp 9m to 1m. 
[image] what do you think ?

Mikespeed2023, 2023-09-04:
Zhank you fpr helpinh , since araising the rf power , rigt now ther was no problem.

utna666, 2023-09-05:
I have the same Problem. RF Power is on max. it worked the whole day and now it cannot connect to the Hoymiles 2000 (2x).
I always get the Error:

17:48:51.175 > RX Period End
17:48:51.175 > All missing
17:48:51.175 > Nothing received, resend count exeeded
17:48:51.223 > TX ActivePowerControl 865.00 MHz --> 51 81 03 61 96 80 10 10 52 81 0B 00 15 4A 00 00 BA 24 BD 

Any help?