tbnobody/OpenDTU #1313: Lost Connection to Hoymiles 2000

utna666, 2023-09-05:
### What happened?

I have 2 Hoymiles 2000 Inverter installed and they worked fine but today the OpenDTU lost Connection to both Inverters and now my Limit is set to 0%.

Both Inverters worked fine for the last 2 Days but now none of them are working anymore.

### To Reproduce Bug

dont know

### Expected Behavior

dont know

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.8.22

### Relevant log/trace output

```Shell
17:58:31.091 > RX Period End
17:58:31.091 > All missing
17:58:31.091 > Nothing received, resend count exeeded
17:58:31.234 > TX ActivePowerControl 865.00 MHz --> 51 74 90 43 89 80 10 10 52 81 0B 00 06 D6 00 00 10 E0 07 
17:58:31.290 > Interrupt received
17:58:31.336 > RX 865.00 MHz --> D1 74 90 43 89 80 10 10 52 81 00 00 0B 00 14 07 B4 | -63 dBm
17:58:34.496 > RX Period End
17:58:34.496 > Success
17:58:34.635 > TX ActivePowerControl 865.00 MHz --> 51 74 90 43 89 80 10 10 52 81 0B 00 09 92 00 00 11 A3 0E 
17:58:36.001 > Interrupt received
17:58:36.146 > RX 865.00 MHz --> D1 74 90 43 89 80 10 10 52 81 00 00 0B 00 14 07 B4 | -62 dBm
17:58:36.195 > RX Period End
17:58:36.195 > Success
17:58:36.258 > TX ActivePowerControl 865.00 MHz --> 51 81 03 61 96 80 10 10 52 81 0B 00 0E B0 00 00 6F 02 AF
```

tbnobody, 2023-09-05:
But your log shows that the inverter gets a successfull ActivePowerControl  command.... But every 5 seconds is way to fast.... if you flood the command queue with packages no stats request will be sent.

tbnobody, 2023-09-05:
> ive set it to 10 sec now but same error

Much to fast... Try 30 seconds at least....

utna666, 2023-09-05:
Ok i will try and send u tomorrow what has happend. Ty for your help

broth-itk, 2023-10-26:
Why is it even possible to set such low values without a warning or note of any type?

tbnobody, 2024-03-21:
Closed due to inactivity

closed 2024-03-21 by tbnobody