tbnobody/OpenDTU #1314: Web interface shows wrong limit values ​​- factor 1000 too small

huste511, 2023-09-05:
### What happened?

Two HMS-2000 are stored in the DTU. Operation during the day without abnormalities. In the evening, however, the web interface shows a limit setting of 0.10% or 2 watts.
No limit setting was ever made.
It affects both MWR.

[image]
[image]

### To Reproduce Bug

new ESP32 deleted and flashed with pre-compiled binaries. Then restored config.json and pin mapping.json.

No abnormalities in operation.

The two json come from an identical ESP32 with the same pre-compiled binaries.

Both MWR started the inverter again in the evening, so that no daily values ​​are stored. This happens frequently, but otherwise no limitations are affected.

Limit settings were never made in the json files or in the current ESP32.

### Expected Behavior

no limit change or display the limit incorrectly.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.8.28

tbnobody, 2023-09-05:
I have a HMS-2000 in front of me. Limit settings works perfectly fine. Did you set a permanent value before which was applied after the power cycle?

huste511, 2023-09-05:
No, I have never touched limit settings. Not even saved the default values ​​again.

I would like to rule out errors caused by an accidental click, since both are affected.

tbnobody, 2023-09-06:
Please provide the output of `http://IP/api/limit/status` and most important, goto "Info" --> "Console" and wait until you see `SystemConfigPara` and then provide several lines around this output.

huste511, 2023-09-06:
Unfortunately, this request comes a little late. The MWR are currently running and real values ​​can be seen.

Output from api/limit/status therefore looks consistent.

`{"1234":{"limit_relative":100,"max_power":2000,"limit_set_status":"Ok"},"4321":{"limit_relative":100,"max_power":2000,"limit_set_status":"Ok"}}`

But I have a new observation!

I have not restarted the MWR or the DTU. I haven't made any settings either.

The 0.1% could therefore be seen until this morning. When there was some light, both MWRs jumped on briefly and showed normal values. Data then went blank again for a while and the reading has changed back to 0.1%.
A few minutes later the MWR were back and the display has been correct ever since.

tbnobody, 2023-09-06:
>  A few minutes later the MWR were back and the display has been correct ever since.

In this case I would assume that the inverter answers with incorrect values if it has just low power. (maybe related to the firmware in the inverter itself?) To debug this, as mentioned above a few lines of the console (with `SystemConfigPara`) are required. (in case of wrong values and in case of correct values)

huste511, 2023-09-06:
Um welchen gesendeten Befehl oder empfange Antwort geht es dann genau?
Am Handy rennt die Console sekündlich durch. Am Rechner, gleich in zwei verschiedenen Browsern, gibt es große Lücken von 5-10 Sekunden. Keine Plan ob dann alles dabei ist.
Daten von den MWR kommen mit ca. -50dBm rein. Wifi steht mit 80 Mbit/s.
Kein Pingverlust aber:
`    Pakete: Gesendet = 160, Empfangen = 160, Verloren = 0
    (0% Verlust),
Ca. Zeitangaben in Millisek.:
    Minimum = 3ms, Maximum = 254ms, Mittelwert = 107ms`
Normal für den ESP32 mit openDTU?

tbnobody, 2023-09-06:
> Um welchen gesendeten Befehl oder empfange Antwort geht es dann genau?

`SystemConfigPara`

huste511, 2023-09-07:
Had the ESP32 without power for a short time today due to antenna replacement. But even with the reboot, you could see the same unusual performance limitations at sunset.

If the MWR no longer feeds in power but sends data, everything is still ok.

[image]

If he doesn't deliver any more data, everything is "normal".

Without clouds and with constant light fading at sunset, the MWR start again 5-10 minutes later. (It's like this almost every evening ;-( )
[image]
(btw: Begi*nn*)

From then on, the information no longer seems correct. (note event counter changes to 1)

[image]

In OpenDTU/api/limit/status , the absolute performance value is correctly stored but not the relative one:

`{"1234":{"limit_relative":0.100000001,"max_power":2000,"limit_set_status":"Ok"},
"4321":{"limit_relative":0.100000001,"max_power":2000,"limit_set_status":"Ok"}}`

Here is the hardware and software version of the MWR, these are identical.
[image]

I have SystemConfig Para in various variations from the time *roughly* of inverter restart or off, unfortunately all without RX:

```
TX SystemConfigPara 865.00 MHz --> 15 91 43 79 59 80 13 52 36 80 05 00 64 F8 C3 1C 00 00 00 00 00 00 00 00 DB 9B 96

TX SystemConfigPara 865.00 MHz --> 15 91 43 75 53 80 13 52 36 80 05 00 64 F8 C3 20 00 00 00 00 00 00 00 00 DA 9A AC

TX SystemConfigPara 865.00 MHz --> 15 91 43 79 59 80 13 52 36 80 05 00 64 F8 C3 24 00 00 00 00 00 00 00 00 1A A8 5C

TX SystemConfigPara 865.00 MHz --> 15 91 43 75 53 80 13 52 36 80 05 00 64 F8 C3 2B 00 00 00 00 00 00 00 00 EA E9 E4

TX SystemConfigPara 865.00 MHz --> 15 91 43 75 53 80 13 52 36 80 05 00 64 F8 C3 31 00 00 00 00 00 00 00 00 8A 5A 2D
```

Log before that unfortunately without SystemConfig Para. I can deliver values ​​during the day.

But if I understand it correctly, it's about the exact moment when the limit values ​​in the GUI change?

Unfortunately, the console here is too incomplete - no idea why.

where is this input and output broken down and explained? Google finds almost nothing.

huste511, 2023-09-07:
Here is a log during normal operation:
```
07:56:26.823 > Fetch inverter: 1234
07:56:26.823 > Request SystemConfigPara
07:56:26.885 > TX RealTimeRunData 865.00 MHz --> 15 91 43 75 53 80 13 52 36 80 0B 00 64 F9 66 0A 00 00 00 00 00 00 00 00 1B 60 17 
07:56:27.033 > Interrupt received
07:56:27.232 > RX 865.00 MHz --> 95 91 43 75 53 80 13 52 36 01 00 01 01 7A 01 86 00 1C 00 1A 00 6C 00 68 00 00 68 | -51 dBm
07:56:27.443 > Interrupt received
07:56:27.501 > RX 865.00 MHz --> 95 91 43 75 53 80 13 52 36 02 62 CF 00 00 6C F5 00 07 00 06 01 89 00 06 00 18 37 | -51 dBm
07:56:27.560 > Interrupt received
07:56:27.648 > RX 865.00 MHz --> 95 91 43 75 53 80 13 52 36 03 00 00 00 60 00 00 00 00 61 DB 00 00 39 8F 00 06 FF | -51 dBm
07:56:27.698 > Interrupt received
07:56:27.845 > RX 865.00 MHz --> 95 91 43 75 53 80 13 52 36 04 00 00 09 26 13 89 01 25 00 00 00 0C 03 E7 00 84 6F | -51 dBm
07:56:27.895 > Interrupt received
07:56:27.944 > RX 865.00 MHz --> 95 91 43 75 53 80 13 52 36 85 00 03 61 EE 9F | -51 dBm
07:56:28.005 > RX Period End
07:56:28.005 > Success
07:56:28.067 > TX SystemConfigPara 865.00 MHz --> 15 91 43 75 53 80 13 52 36 80 05 00 64 F9 66 0A 00 00 00 00 00 00 00 00 D5 6F D8 
07:56:28.255 > Interrupt received
07:56:28.302 > RX 865.00 MHz --> 95 91 43 75 53 80 13 52 36 81 00 01 03 E8 00 00 03 E8 00 00 00 00 00 00 3C F8 D2 | -51 dBm
07:56:28.457 > RX Period End
07:56:28.457 > Success
07:56:28.666 > Fetch inverter: 4321
```

At the start, it switched between 0.1% and 100% several times. Unfortunately again no complete log

tbnobody, 2023-09-07:
> 95 91 43 75 53 80 13 52 36 81 00 01 03 E8 00 00 03 E8 00 00 00 00 00 00 3C F8 D2

The relevant Bytes are `03 E8` Which have to be converted to decimal (`1000`) and divided by `10` --> `100`%

I would assume the inverter sends some unrealistic value when it has low power. This is nothing which can be corrected by the DTU because it doesn't know that the values are wrong. (CRC checksums etc are correct in this place)

huste511, 2023-09-07:
MWR responses for real-time data of the strings and SystemConfigPara always consist of 27 “hex packets”?
Real-time data from the MWR has 15 “hex packets”?

Then I think I have something in two places, around the time where the wrong limit was displayed:

```
20:19:15.145 > TX RealTimeRunData 865.00 MHz --> 15 91 43 79 59 80 13 52 36 80 0B 00 64 FA 14 22 00 00 00 00 00 00 00 00 BF 32 BE 
20:19:15.199 > Interrupt received
20:19:15.352 > RX 865.00 MHz --> 95 91 43 79 59 80 13 52 36 81 0B 00 09 00 50 04 47 | -52 dBm
20:19:15.416 > RX Period End
20:19:15.416 > ERROR in RealTimeRunData: Received fragment size: 6, min expected size: 66
20:19:15.416 > Packet handling error
20:19:15.473 > TX SystemConfigPara 865.00 MHz --> 15 91 43 79 59 80 13 52 36 80 05 00 64 FA 14 22 00 00 00 00 00 00 00 00 71 3D 71 
20:19:15.530 > Interrupt received
20:19:15.657 > RX 865.00 MHz --> 95 91 43 79 59 80 13 52 36 81 05 00 00 01 28 C1 FC | -52 dBm
20:19:15.761 > RX Period End
20:19:15.761 > Success
```

```
20:23:22.497 > TX RealTimeRunData 865.00 MHz --> 15 91 43 79 59 80 13 52 36 80 0B 00 64 FA 15 19 00 00 00 00 00 00 00 00 4B 44 06 
20:23:22.564 > Interrupt received
20:23:22.619 > RX 865.00 MHz --> 95 91 43 79 59 80 13 52 36 81 0B 00 09 00 50 04 47 | -52 dBm
20:23:22.801 > RX Period End
20:23:22.801 > ERROR in RealTimeRunData: Received fragment size: 6, min expected size: 66
20:23:22.801 > Packet handling error
20:23:22.856 > TX SystemConfigPara 865.00 MHz --> 15 91 43 79 59 80 13 52 36 80 05 00 64 FA 15 19 00 00 00 00 00 00 00 00 85 4B C9 
20:23:22.945 > Interrupt received
20:23:22.995 > RX 865.00 MHz --> 95 91 43 79 59 80 13 52 36 81 05 00 00 01 28 C1 FC | -52 dBm
20:23:23.146 > RX Period End
20:23:23.146 > Success
```

The 13+14 digits are dacimal only "1", which corresponds to the incorrect performance information.

I can't understand the size 6 or 66.

An error is detected, but the complaint no longer occurs in the second attempt?

What is being mistransmitted or misinterpreted here?

How does the original DTU behave in such a case?

The reasons can be found in the HMS-2000 or rather in my (old?) firmware? Others must have noticed it since it's been there all night.

tbnobody, 2023-09-07:
> I can't understand the size 6 or 66. 

That's the received data: `95 91 43 79 59 80 13 52 36 81 0B 00 09 00 50 04 47` 
The important payload is `0B 00 09 00 50 04` but at least 66 bytes are expected. the inverter only transmitted 6 so the package is dropped.

> 20:23:22.995 > RX 865.00 MHz --> 95 91 43 79 59 80 13 52 36 81 05 00 00 01 28 C1 FC | -52 dBm

In this case, the payload looks also too short compared to the good one (see above)

Corrupt:
`95 91 43 79 59 80 13 52 36 81 05 00 00 01 28 C1 FC` --> Important payload `00 01` --> 1 / 10 = 0.1%

Good one:
`95 91 43 75 53 80 13 52 36 81 00 01 03 E8 00 00 03 E8 00 00 00 00 00 00 3C F8 D2` --> important  payload `03 E8` --> 1000 / 10 = 100%

But that means I have to add a filter in the SystemConfigPara parser as well to get rid of this corrupt packages (the packages have a correct CRC checksum so it's not a RF issue but a wrong calculation in the inverter)

tbnobody, 2023-09-11:
Should be fixed in 8023b6620aa0867773a284c961fb8c3e4bd902d0

closed 2023-09-11 by tbnobody