tbnobody/OpenDTU #1316: Error opening webinterface

TrZiegler, 2023-09-07:
### What happened?

After installing the last release v23.8.28 opening of the webinterface ist prolonged, an error is reported in the log:
Error 2, DTU command failed. The noted time is wrong-in the future! Reading the error-log again, time is correct ! 

### To Reproduce Bug

Seems to occur after the start of the inverter in the morning. Further request are then working without a problem.  

### Expected Behavior

Startup without any problem as before update!

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.8.28

### Relevant log/trace output

```Shell
06:40:08	06:40:08	1	Inverter start
10:26:34	10:26:34	2	DTU command failed
10:27:04	10:27:04	2	DTU command failed
```

TrZiegler, 2023-09-07:
now I am using Version v23.8.22, problem does not recurr until now !

tbnobody, 2023-09-07:
Version v23.8.22 didn't change anything related to the eventlog or times etc. (https://github.com/tbnobody/OpenDTU/compare/v23.8.22...v23.8.28)

The topic with the `DTU command failed` entry should be fixed in one of the future releases. It happens when the inverter receives the `SystemConfigPara` request but the DTU does not receive the answer and re-requests the package.

tbnobody, 2024-01-13:
Going to close this issue. Please feel free to reopen or create a new issue if the problem persists.

closed 2024-01-13 by tbnobody