tbnobody/OpenDTU #1319: Connect to WiFi using DHCP not working

nobodyman1, 2023-09-08:
### What happened?

Using the latest version v23.8.28 it´s not possible to connect to WiFi network.

### To Reproduce Bug

Take a blank ESP32 module and flash opendtu-generic.factory.bin.
Connect to the local access point "OpenDTU-*" and configure WLAN-SSID and WLAN-Password and confirm settings.
After a short time connect to local access point again and check "Info" - "Network".
All IP addresses remain at 0.0.0.0.

### Expected Behavior

I expect valid IP settings.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.8.28

### Anything else?

I cleaned the whole device and tried different versions.

| Version | DHCP |
| --- | --- |
| v23.8.22 | working |
| v23.8.8 | not working |
| v23.8.3 | working |

tbnobody, 2023-09-08:
did you reset after the update or did you perform a power cycle? I did not change anything regarding wifi in any of these versions.

nobodyman1, 2023-09-09:
I checked it again. If DHCP is not working an additional power cycle is needed.
It might be useful to add this to your documentation.

tbnobody, 2023-09-10:
Done... going to close this issue.

closed 2023-09-10 by tbnobody