tbnobody/OpenDTU #1321: No connection to HM-1500

BennyOtt, 2023-09-09:
### What happened?

After installing the openDTU, I tried to establish a connection to the inverter. A connection to my WiFi is perfectly fine, but not to the inverter. The corresponding serial number is entered, the correct product type is also displayed, but there is no live data, although the system is producing

### To Reproduce Bug

The following is displayed accordingly in the log
[image]

### Expected Behavior

that the live data is displayed accordingly and the logs do not say that no data is being received

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.8.28

### Relevant log/trace output

```Shell
17:05:43.211 > All missing
17:05:43.211 > Nothing received, resend whole request
17:05:43.211 > TX AlarmData Channel: 75 --> 15 90 74 13 53 80 18 56 40 80 11 00 64 FC 89 C1 00 00 00 00 00 00 00 00 5E 7D 5D 
17:05:44.028 > RX Period End
17:05:44.028 > All missing
17:05:44.028 > Nothing received, resend whole request
17:05:44.028 > TX AlarmData Channel: 3 --> 15 90 74 13 53 80 18 56 40 80 11 00 64 FC 89 C1 00 00 00 00 00 00 00 00 5E 7D 5D 
17:05:44.780 > RX Period End
17:05:44.780 > All missing
17:05:44.780 > Nothing received, resend count exeeded
17:05:44.888 > TX SystemConfigPara Channel: 23 --> 15 90 74 13 53 80 18 56 40 80 05 00 64 FC 89 C1 00 00 00 00 00 00 00 00 4A 69 49 
17:05:45.197 > RX Period End
17:05:45.197 > All missing
17:05:45.197 > Nothing received, resend whole request
17:05:45.197 > TX SystemConfigPara Channel: 40 --> 15 90 74 13 53 80 18 56 40 80 05 00 64 FC 89 C1 00 00 00 00 00 00 00 00 4A 69 49 
17:05:45.508 > RX Period End
17:05:45.508 > All missing
17:05:45.508 > Nothing received, resend whole request
17:05:45.508 > TX SystemConfigPara Channel: 61 --> 15 90 74 13 53 80 18 56 40 80 05 00 64 FC 89 C1 00 00 00 00 00 00 00 00 4A 69 49 
17:05:45.821 > RX Period End
17:05:45.821 > All missing
17:05:45.821 > Nothing received, resend whole request
17:05:45.821 > TX SystemConfigPara Channel: 75 --> 15 90 74 13 53 80 18 56 40 80 05 00 64 FC 89 C1 00 00 00 00 00 00 00 00 4A 69 49 
17:05:46.128 > RX Period End
17:05:46.128 > All missing
17:05:46.128 > Nothing received, resend whole request
17:05:46.128 > TX SystemConfigPara Channel: 3 --> 15 90 74 13 53 80 18 56 40 80 05 00 64 FC 89 C1 00 00 00 00 00 00 00 00 4A 69 49 
17:05:46.333 > RX Period End
17:05:46.333 > All missing
17:05:46.333 > Nothing received, resend count exeeded
17:05:46.333 > Fetch inverter: 116190......
17:05:46.333 > Request SystemConfigPara
17:05:46.434 > TX RealTimeRunData Channel: 23 --> 15 90 74 13 53 80 18 56 40 80 0B 00 64 FC 89 CA 00 00 00 00 00 00 00 00 B4 15 CE 
17:05:47.046 > RX Period End
17:05:47.046 > All missing
17:05:47.046 > Nothing received, resend whole request
17:05:47.046 > TX RealTimeRunData Channel: 40 --> 15 90 74 13 53 80 18 56 40 80 0B 00 64 FC 89 CA 00 00 00 00 00 00 00 00 B4 15 CE 
17:05:47.656 > RX Period End
17:05:47.656 > All missing
17:05:47.656 > Nothing received, resend whole request
17:05:47.656 > TX RealTimeRunData Channel: 61 --> 15 90 74 13 53 80 18 56 40 80 0B 00 64 FC 89 CA 00 00 00 00 00 00 00 00 B4 15 CE 
17:05:48.263 > RX Period End
17:05:48.263 > All missing
17:05:48.263 > Nothing received, resend whole request
17:05:48.263 > TX RealTimeRunData Channel: 75 --> 15 90 74 13 53 80 18 56 40 80 0B 00 64 FC 89 CA 00 00 00 00 00 00 00 00 B4 15 CE 
17:05:48.875 > RX Period End
17:05:48.875 > All missing
17:05:48.875 > Nothing received, resend whole request
17:05:48.875 > TX RealTimeRunData Channel: 3 --> 15 90 74 13 53 80 18 56 40 80 0B 00 64 FC 89 CA 00 00 00 00 00 00 00 00 B4 15 CE
```

tbnobody, 2023-09-09:
What shows the radio information in "Info" --> "System"?

BennyOtt, 2023-09-09:
> What shows the radio information in "Info" --> "System"?

That everything is connected and configured correctly.
[image]

tbnobody, 2023-09-10:
Please doublecheck the IRQ pin connection. If it is not connected it's not shown in the WEBui but no packages are received.
What kind of nRF module are you using?

closed 2023-09-13 by BennyOtt