tbnobody/OpenDTU #1324: [Request] Support for HMS-XXXW-2T

Commifreak, 2023-09-11:
### Describe the solution you'd like

I dont know if this was already discussed, but is any support for the Wifi-integrated models possible? I have a HMS-800W-2T here, which includes WiFi already.

tbnobody, 2023-09-11:
I am nearly 100% certain that wifi devices will never be a use case for this project as it is a bridge between proprietary rf protocols (CMT module or nRF24 module) towards wifi. 
Due to the fact that the HMS-XXXW-2T inverters are already communicating using TCP, all the protocol implementation done in this project are not required. On the other hand, it requires effort to sniff and analyze all the TCP pakets from the inverter to the hoymiles cloud. (which should be nearly impossible if it is SSL encrypted). In every case this would be a pure software project and does not need any additional hardware like ESPs etc....

closed 2023-09-11 by Commifreak

stefan123t, 2023-10-15:
HMS-WiFi Models use the S-Miles Cloud protocol to communicate over the built-in DTUBI WiFi interface. The commands are so called NetworkCommands wrapping NRF/CMT like Gen2/Gen3 codes which we already support. Support for NetworkCommands and DTU-BI will be analyzed as part of #1430 and https://github.com/lumapu/ahoy/issues/1189

gitisgreat2023, 2024-01-10:
FYI, first implementation can be found [here](https://github.com/DennisOSRM/hms-mqtt-publisher/tree/main).