tbnobody/OpenDTU #1325: Dynamic Limit API

unaiur, 2023-09-11:
### Is your feature request related to a problem? Please describe.

I want to implement dynamic power limits using Web API of OpenDTU to ensure we do not dump more than 600w to the network as required in most countries in EU.

The problem is that, if the WIFI or the controller dies, the limit may be left unmanaged at a value that is too high.

### Describe the solution you'd like

The easiest solution I can envision is a new dynamic limit WebAPI that receives a temporal limit in Watts and a timeout in seconds. This API will cancel any previous scheduled reset that may be pending, apply the new temporal limit as non-persistant and schedule a limit reset in the given amount of seconds.

This way, if no new limit is received in X seconds, we revert to the persistent limit. This will reduce the chances of uncompliant energy dumps greatly.

unaiur, 2023-09-11:
If there is no easy way to retrieve the persistent limit, I would add an extra parameter to the new API with the fallback limit to apply when the timer expires.

Heiko-L, 2023-09-12:
> I want to implement dynamic power limits using Web API of OpenDTU to ensure we do not dump more than 600w to the network as required in most countries in EU.

I think you don't understand the intention to the (8/6)00W limit.

Why not set the persistent limit to 600W and change only non persistent afterwards?

unaiur, 2023-09-15:
I want to implement some logic that every second measures the power consumed in my home and limits the panels output to the consumed power + 400W.

Suppose that I turn on my 900W microwave; the limit is raised to 1300W but some seconds later my wifi hangs and no updates are sent anymore. One minute later my coffee is hot, the microwave stops and I am dumping 1300W into the electricity network, way above the 600W limit, and it will continue doing so while the weather is sunny and the WiFi is down.

I would like to reduce the likelihood of that scenario.

unaiur, 2023-09-16:
Of course, the persistent limit is set at 600W. I fully agree on that and now I have permanently set at this limit. 

But, when my home consumption is above 600W and no power is dumped on the grid, I would like to raise the limit to my current consumption. 

For example, if I turn on the oven (which consumes ~2000W), I would like to temporarily raise the limit to 1200 (maximum for my panels). If I turn on the microwave (~900W), I would like to raise the limit to 900W.

No problem here. I have an energy meter that can provide measures multiple times per second and can raise the limit when possible.

The problem is reducing the limit when the oven or microwave finishes.

If everything goes well, the limit will be readjusted in milliseconds and there is no problem at all.

But imagine something happens to my wifi and the command to reduce the limit never reach OpenDTU.

That is the problem I would like to avoid having OpenDTU rolling back to the permanent limit if the temporal limit is not updated in last X seconds.

Of course, it will be even better to have this logic in the inverter, but its firmware is not open source.

jonnycastaway, 2023-10-05:
Ok, but the easy way is, read the actual limit regulary and if it's not as expected set the limit again.
So you can cover small wifi outaged and it's best practice as well. Never trust an fire&forget.
If your wifi outage is longer you have to fix your wifi.

spcqike, 2024-02-06:
> But imagine something happens to my wifi and the command to reduce the limit never reach OpenDTU.

how do you deal with problems on your openDTU itself? 

You can always have problems somewhere. If it’s not WiFi, it’s the power brick of your DTU or the DTU itself. If you need a real and permanent solution, you need some other hardware. 

In my case my server and wifi is way more reliable than my openDTU or the NRF-connection.

stefan123t, 2024-08-06:
@unaiur please double check the **Dynamic Power Limiter (DPL)** function in the daughter project[ OpenDTU-OnBattery](https://github.com/helgeerbe/OpenDTU-OnBattery/).

I have described a [possible PM+DPL setup](https://github.com/tbnobody/OpenDTU/issues/272#issuecomment-2219644352) with a Shelly Pro3EM as **Power Meter (PM)** here in issue #272.

jonnycastaway, 2024-08-06:
> @unaiur please double check the **Dynamic Power Limiter (DPL)** function in the daughter project[ OpenDTU-OnBattery](https://github.com/helgeerbe/OpenDTU-OnBattery/).
>
>
>
> I have described a [possible [...]

But OpenDTU-OnBattery can only use 1 Inverter for this function.
With an extra System you can regulate as many as you want(4 in my case).

stefan123t, 2024-08-06:
@jonnycastaway i am not sure whether OpenDTU-OnBattery only limits a single inverter. I thought DPL would limit all inverters, but then I have only one and cannot test / verify that.

Some made a suggestion to group inverters into PV pools, eg per phase A/B/C or per battery. That way you could specify a per pool limit or set the limits based on a schedule (before/after sunset).

See feature request #1865 and discussions #1608 and #1095.

jonnycastaway, 2024-08-07:
@stefan123t At the moment you can read multiple Inverters but for limitation you can only use one.
Yes there are ideas with groups but as i can see nothing is implemented.
Setting the limit of a group is not an option for me because is have 4 systems an they look at different directions. So why limit a system that generates only little power.
My logic is doing this every 10 seconds:
Read the generated power (W), select the system that generates the most power, decrease this system.
For increasing it selects the system with the lowest limit.

But'll have an eye on -OnBattery because it would very cool to have this function on the DTU. In the meantime i'm working on to migrate my logic from javascript/blockly to an extra ESP32 with display and to visualize the systems.

[image]

stefan123t, 2024-08-15:
@jonnycastaway I understood from @schlimmchen that he is working on an improved DPL feature in OpenDTU-OnBattery with the purpose of managing / controlling more than one inverter.