tbnobody/OpenDTU #1333: Live-Daten only shows white screen

seisfeld, 2023-09-12:
### What happened?

Live-Daten does not work anymore after latest [v23.9.11](https://github.com/tbnobody/OpenDTU/releases/tag/v23.9.11) update. Clicking on Live-Daten only produces a white screen the menu header is visible though. Also the rest of OpenDTU seems operational. It reports values via MQTT, inverter is reachable and producing etc. nothing out of the ordinary... 

### To Reproduce Bug

Do the [v23.9.11](https://github.com/tbnobody/OpenDTU/releases/tag/v23.9.11) update and click on the Live-Daten menu option.

### Expected Behavior

Live-Daten should show data and not a white screen.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.9.11

### Relevant log/trace output

```Shell
The browser (Safari int this case, bit I also tried Chrome) shows the following in the java script console:

[Error] WebSocket connection to 'ws://192.168.10.56/console' failed: The operation couldn’t be completed. (kNWErrorDomainPOSIX error 54 - Connection reset by peer)
[Error] WebSocket connection to 'ws://192.168.10.56/livedata' failed: The operation couldn’t be completed. (kNWErrorDomainPOSIX error 54 - Connection reset by peer)
[Error] TypeError: null is not an object (evaluating 'this.liveData.inverters')
	inverterData (app.js:59:51033)
	inverterData
	run (app.js:2:5023)
	value (app.js:2:12841)
	get (app.js:2:23538)
	get (app.js:2:21772)
	(anonyme Funktion) (app.js:59:51709)
[Error] WebSocket connection to 'ws://192.168.10.56/livedata' failed: The operation couldn’t be completed. (kNWErrorDomainPOSIX error 54 - Connection reset by peer) (x5)
```

tbnobody, 2023-09-12:
Did you reload the web interface or even better did you clear the cache?

seisfeld, 2023-09-12:
Yes. I always force reload the page as indicated after the upgrade. Followed your suggestion to clear the cache on both browsers, no dice.

The console is working fine. It shows that it's fetching the inverter and it's results. It's just the Live-Daten thing which is dead.

seisfeld, 2023-09-12:
I power cycled OpenDTU now, it seems to work now. I still get these messages in the js console occasionally though. I don't know if these where there before the upgrade or not. Will monitor and will report again if I can reproduce the white screen somehow.

> [Error] WebSocket connection to 'ws://192.168.10.56/livedata' failed: The operation couldn’t be completed. (kNWErrorDomainPOSIX error 54 - Connection reset by peer)

tbnobody, 2023-09-12:
Background: Due to the limited memory on the ESP32 only one resource intensive web request will be handled at a time. e.g. when loading the live view, the initial data is fetched via the `/api/livedata/status` get request. Shortly after that, the web socket connection is established. If this happens too fast (or memory from the get request is not yet released) the web socket returns null (and gets disconnected). After 10 seconds the frontend retries the web socket connection (which now succeeds). Maybe that's what you see in the Safari log. I was just wondering why you got such exceptions in the first place because I added several patches to handle this null value. (and therefor I asked whether you've reloaded the frontend)

seisfeld, 2023-09-12:
Thanks for the explanation and your work in general! It's the first time I experienced such issue after an upgrade. So maybe I did hit some weird edge case. I immediately used Chrome to verify (I never actiually opened the webif in Chrome before because I mainly use Safari for day to day use - so this rules out a caching issue IMHO) and when I saw it not working there either I thought something might not be right. But somehow a power cycle fixed it now... it works perfectly fine again.

closed 2023-09-13 by tbnobody