tbnobody/OpenDTU #1339: AZDelivery ESP32 D1 Mini NodeMCU won't boot with v23.9.11 or newer

puni2k, 2023-09-14:
### What happened?

Updated my AZDelivery ESP32 D1 Mini NodeMCU from v23.8.28 to v23.9.13.

openDTU wouldn't boot after this. Fresh install with Espressif Flash shows same symptoms.
Same problem also with v23.9.11

After flashing with v23.8.28 openDTU works again.

Serial output looks kinda strange (see log section)
Don't think this is a power supply issues since it happens whether the ESP is connected to a PC or a dedicated charger with plenty of wattage (2.4W) and only represents itself on the two latest firmwares.

### To Reproduce Bug

Flash specified ESP with v23.9.11 or newer.

### Expected Behavior

ESP should not hang in a boot loop

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.8.28

### Relevant log/trace output

```Shell
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (631) esp_core_dump_flash: No core dump▒▒ѥѥ▒▒▒found!
E (632) esp_core_dump_flash: No core dump partition found!

Starting OpenDTU
Initialize FS... done
Reading configuration... done
Reading PinMapping... [   155][E][vfs_api.cpp:105] open(): /littlefs/pin_mapping.json does not exist, no permits for creation
using default config done
Initialize Network...
Brownout detector was triggered
```

puni2k, 2023-09-14:
Maybe related
https://github.com/tbnobody/OpenDTU/discussions/1331

tbnobody, 2023-09-14:
The message "Brownout detector was triggered" does not come from OpenDTU. Its out of the Espressiff Core. If you think that this is a issue, then its not a OpenDTU one. I cannot even influence the Brown Out detector.

turrican944, 2023-09-14:
I have the same ESP. It works without any problems with version 23.9.13, but I use my own
compiled binary.
Maybe it's the ESP problem or power supply
[image]

puni2k, 2023-09-14:
Figured that much :(

Tried out 3 different power supplies including one from an ipad and multiple USB ports on my PC / Display and multiple USB cables.
All with the same result.

@turrican944 i noticed that you have quite a big cap for the NRF24L01+ so i guessed that also might help to stabilize the ESP. I added an extra 100µF to mine but that didn't help either.
Only have a simple multimeter to check the voltage. Stays around 3.25V the whole time but it might be to slow.

Tried another AZDelivery ESP (not a D1 Mini) I keep as a backup system, which works with the power supply / cable combination, so i guess there might be some kind of incompability between my D1 Mini and the new ESP firmware.

I'm fine with closing this issue, but I wanted to have it documented in case someone else runs into the same problem.

closed 2023-09-14 by puni2k

puni2k, 2023-09-15:
Just a follow up.
I managed to make my D1 Mini work with the newer firmwares after doing a bit of reading on ESP dev boards.

Seems like that the power curcuit on some of them isn't adequately deisgned to fully comply with the USB specification. Most of the voltage regulators should be able to safely operate even if the power supply is at the low end of the USB spec (if you don't have a huge voltage drop on the cable).
However most boards but a diode between USB V+ and the voltage regulator input so you have significant voltage drop here (in my case 0.3V).
So even if you have a power supply that operates within spec and low voltage drop over the cable the resulting voltage on the regulator might not be sufficient for reliable operation.

If you have the option, a power supply that operates towards the upper limit of the USB specification (5.5V if i remember correctly) would be best.

I bridged the diode in question and added a 47µF cap between USB V+ and ground to be on the safe side. Works for now. The disadvantage with this is that you can no longer connect the board via USB while it is powered by a separate 5V source. Not a problem for my use case though.