tbnobody/OpenDTU #1341: [Security issue] Hide serial number of inverter for anonymous users

pkese, 2023-09-14:
### Is your feature request related to a problem? Please describe.

I've made my OpenDTU status web page publicly visible so that I can see the performance without signing in.

The problem is that this makes the serial numbers of inverters exposed to public, meaning anyone with a NRF24L01 board can now come to my house, copy&paste the serial numbers of the inverters from the OpenDTU status page and fiddle with my inverters.

### Describe the solution you'd like

I'd propose not to display serial numbers of inverters to anonymous users  
and only to make them visible to logged-in users.

### Describe alternatives you've considered

Disabling "readonly access to web interface without password", but then ... why having this option in the first place.

### Additional context

Each inverter can be assigned a custom name anyway, which should suffice for identification.

tbnobody, 2023-09-14:
1. I wrote several times that users never ever should expose iot devices like this with port forwarding 
2. If you are using a reverse proxy with ssl certificate you can still disable the anoymous access
3. I always mentioned to use a VPN like OpenVPN or Wireguard
4. If I really know where your house lives I can just use a NRF24L01 board (or any other RF sniffer) and sniff the sent packages direct out of the air. No need to know any address of your DTU.

> Disabling "readonly access to web interface without password", but then ... why having this option in the first place.

To grant your children (or anyone else in your wifi) a look at the current production without letting them change any settings

> Each inverter can be assigned a custom name anyway, which should suffice for identification.

A pure number is something completly different to something which the user can define (e.g. utf8 charsets, escaping etc.)

stefan123t, 2024-08-15:
While I understand the criticism for exposing the OpenDTU to the public (implies Internet) !

I also understand the request from several users to have some more "privacy" in the OpenDTU UI, i.e. the last five digits of the inverter shown on the read-only pages or the Live homepage may have to be obscured / anonymized using an additional image editing process. Also the full inverter serial ID shown in the Serial Console logs have to be replaced using manual search&replace before posting them as evidence to an issue. IMHO the prefix of the inverter #N should be sufficient for distinguishing an issue in the console logs.

While this is definitely not a big issue, it would be nice if this extraneous (maybe superfluous) information is hidden using asterisks in the UI and web console to address users tendency for data scarcity. It may be demasked using an extra button or hovering over for verification, where this may be necessary, i.e. maybe the masking could also be executed in the UI Vue.js component IMHO, as it shouldn't be visible in the first place. And as can be seen from Radio traces the data is out there in the first place anyway.