tbnobody/OpenDTU #1343: Regelung ser Leistung wird erst nach Neustart angenommen

Mille641, 2023-09-15:
### What happened?

Habe ein Smart Micro Solar zur Nullregelung mit eingebunden. Dies funktioniert über ca. 6 Stunden perfekt. Doch dann übermittelt die DTU keine Werte mehr an die Wechselrichter. Nach einem Neustart klappt dies dann wieder. 

### To Reproduce Bug

Neustart der DTU

### Expected Behavior

Dauerhafte Übermittlung der Werte vom Smart Micro Solar 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

V23.8.22

tbnobody, 2023-09-15:
Wie hast du den inverter eingebunden? Setzt du die Limit Werte via MQTT? In welchem Zeitinterval?

Mille641, 2023-09-15:
Mit diesem Gerät ...

https://selbstbau-pv.de/wissensbasis/selbstbau-datenlogger-fuer-hoymiles-mikrowechselrichter-mit-leistungsbegrenzung/

tbnobody, 2023-09-15:
naja, aber du benötigst ja trotzdem ein stück software das die nulleinspeisung regelt. OpenDTU kann ja kein Smart Micro Solar (was auch immer das ist) ansprechen. Du musst also via MQTT oder web requests etwas an den ESP senden.

tbnobody, 2023-09-15:
Vielleicht solltest du dich dann an den Verkäufer des Gerätes wenden?

Mille641, 2023-09-15:
Das Teil spricht ja die DTU an ...
Hatte allerdings teilweisen auch ohne dem Gerät das gleiche Problem die Leistung der Wechselrichter einzustellen. Bei mir laufen aktuell drei HM-1500.

tbnobody, 2023-09-15:
> Das Teil spricht ja die DTU an ...

Wie? Via Web API oder MQTT? In welchem Interval?

> Hatte allerdings teilweisen auch ohne dem Gerät das gleiche Problem die Leistung der Wechselrichter einzustellen.

D.h. du hast das Problem nicht mehr? Was hast du ändert? Was siehst du auf der Konsole wenn es nicht funktioniert?

Mille641, 2023-09-16:
Ne, leider nicht ...
Dachte mir dem Umbau auf den kleinen Modul wird es vielleicht besser. Aber leider blieb das Problem noch.
So wie ich das sehe verwendet das Teil Web API.

https://selbstbau-pv.de/wissensbasis/smart-micro-solar-einrichtung/

Mille641, 2023-09-16:
Tut mir leid, dass ich nur so halbe Antworten geben kann. Leider bin ich mit dem ganzen nicht so firm. 
Bei dem kleinen Modul fügst du einfach die IP von deinem Shelly und der DTU beim Einrichten hinzu. Und mehr ist nicht zu tun. Dann läuft die Sache mit der Nulleinsparung.

stefan123t, 2024-08-15:
@Mille641 das von selbstbau-pv.de zur Verfügung gestellte Modul namens Smart-Micro-Solar (SMS) steuert die OpenDTU.
Es ist im Gegensatz zur OpenDTU m.W. kein Open Source/Hardware sondern sog. Closed Hardware, d.h. Du musst Dich wegen Deinem Problem an den Verkäufer Deines SMS wenden. 
Dieses Modul nutzt entweder die MQTT oder die REST API von OpenDTU, aber ohne Source Code oder genaue Fehlerlogs wird Dir auf Seite OpenDTU niemand weiterhelfen können.

Es gibt die u.a. Lösung mit der Funktion Dynamic Power Limit des Forks OpenDTU-OnBattery für einen Wechselrichter. 
Eventuell kommt auch bald noch eine Erweiterung für mehrere Wechselrichter dazu.

---

Please double check the **Dynamic Power Limiter (DPL)** function in the daughter project[ OpenDTU-OnBattery](https://github.com/helgeerbe/OpenDTU-OnBattery/).

I have described a [possible PM+DPL setup](https://github.com/tbnobody/OpenDTU/issues/272#issuecomment-2219644352) with a Shelly Pro3EM as **Power Meter (PM)** here in issue #272.

tbnobody, 2025-02-07:
This issue has been automatically closed because it has not had recent activity. Feel free to open a new issue if the problem persists.

closed 2025-02-07 by tbnobody