tbnobody/OpenDTU #1344: Unable to connect to WiFi

thampel, 2023-09-15:
### What happened?

From two identical ESP32 devices, both flashed with exactly the same openDTU image, one is working perfectly and the other is not allowing to connect over WiFi at all. The standard WiFi password (openDTU42) does'nt work.

I've tried to erase + flashing it again, tried different flashing methods, even downloaded the *.bin file again, tried different clients, tried different power adapters, unfortunately all with same result: it doesnt work.

### To Reproduce Bug

1. Flash with [23.9.13](https://github.com/tbnobody/OpenDTU/releases/tag/v23.9.13) 
2. Power Off/Power On
3. Find WiFi named "OpenDTU-xxxxx"...
4. Try to connect to this WiFi using default password "openDTU42" => fail

### Expected Behavior

1. Flash with [23.9.13](https://github.com/tbnobody/OpenDTU/releases/tag/v23.9.13) 
2. Power Off/Power On
3. Find WiFi named "OpenDTU-xxxxx"...
4. Try to connect to this WiFi using default password "openDTU42" => working

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

4ce0c21

### Relevant log/trace output

```Shell
N/A
```

DPO99, 2023-09-15:
Did you flashed opendtu for the first time on theese esp32 or did you update via new flash?

thampel, 2023-09-16:
It was a first-time-flash, but also further attempts to erase/reflash did not solve the problem. The WiFi password still doesnt work despite the OpenDTU-1234567 access point being up and running.

DPO99, 2023-09-16:
What about a older dtu version?

thampel, 2023-09-16:
Bingo! that worked!
I've used [v23.9.11](https://github.com/tbnobody/OpenDTU/releases/tag/v23.9.11) and everything is working just fine.

closed 2023-09-16 by thampel