tbnobody/OpenDTU #1348: No Live-Data on the webpage

itkama, 2023-09-17:
### What happened?

Web-Interface shows now Live-Data anymore after upgrading to the latest firmware.

### To Reproduce Bug

Upgrade to latest Firmware Release. Came from 23.8.28.

### Expected Behavior

Web-Interface shows Live-Data.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.9.13

### Relevant log/trace output

```Shell
From the console:
13:26:46.271 > Websocket: [/livedata][37] connect
13:26:51.142 > Websocket: [/livedata][37] disconnect

From the browser console (Chrome):
Uncaught TypeError: Cannot read properties of null (reading 'inverters')
    at Proxy.inverterData (app.js:54:51033)
    at ue.run (app.js:2:5020)
    at get value [as value] (app.js:2:12837)
    at Object.get [as inverterData] (app.js:2:23538)
    at Object.get (app.js:2:21771)
    at app.js:54:51709
```

### Anything else?

MQTT works fine, Display on the DTU itself too.

tbnobody, 2023-09-17:
Please read #1333 

* make sure you reloaded the web interface (ctrl+f5) 
* if the problem persists, perform a power cycle (really unplug the device)

itkama, 2023-09-17:
@tbnobody 
After reloading the web interface with ctrl+f5 I'm getting a white screen only like in #1333. The problem persists even after a power cycle.

tbnobody, 2023-09-17:
whats the output of http://192.168.5.118/api/livedata/status?

Can you provide your config.json file? (please make sure you replace your wifi credentials with xxx - please same amount of characters)

itkama, 2023-09-17:
The output of that link is "null"
I cannot provide you with the config.json, since I can't reach the web interface at all. As I wrote, everything is just a blank white page now.
MQTT is still running though, same for the display.

tbnobody, 2023-09-17:
If nothing more is working, I would suggest you perform a full erase and reflash the esp again using the factory image.

itkama, 2023-09-17:
Using the ESPFlasher I get the following error:
`Using 'COM5' as serial port.
Connecting.................
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WDQ5-V3 (revision 3)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: B0:A7:32:2A:3B:FC
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)
`
I used the "opendtu-generic_esp32.factory.bin", since it is an ESP32. What am I doing wrong?

itkama, 2023-09-17:
Managed to flash the non-factory-image over it and got the Web-Interface back. Still no Live-Data though (or atleast something on the front page).

Dev-Console of Chrome still shows these errors:
`Uncaught TypeError: Cannot read properties of null (reading 'inverters')
    at Proxy.inverterData (app.js:54:51033)
    at ue.run (app.js:2:5020)
    at get value [as value] (app.js:2:12837)
    at Object.get [as inverterData] (app.js:2:23538)
    at Object.get (app.js:2:21771)
    at app.js:54:51709
`
`app.js:54 WebSocket connection to 'ws://admin:AFvQf9ZkEDvGgPixB2lRraPywGYqLz97c6kqZSl2@192.168.5.118/livedata' failed: Close received after close
`

itkama, 2023-09-17:
Okay, I got further now. I set it up step for step.
If I set it up with TLS enabled for MQTT, I can't see any live data.
If I set it up with TLS disabled for MQTT, everything works. So there seems to be some problem with the TLS function of MQTT?

casimirextreme, 2023-09-18:
I have exactly the same issue.  I'm also using TLS enabled MQTT.
OpenDTU compiled by myself (since the beginning that I'm using it). It always worked flawlessly.

casimirextreme, 2023-09-18:
FYI. I went back to v23.8.28 and it's working fine

tbnobody, 2023-09-18:
Please check with current version

casimirextreme, 2023-09-18:
Yes, issue solved on my side. Thanks @tnobody

closed 2023-09-19 by itkama