tbnobody/OpenDTU #1354: [Request] WIFI configuration - allow definition of preferred Access Point MAC address

swingstate, 2023-09-18:
### Is your feature request related to a problem? Please describe.

The DTU I am using is installed next to a wireless repeater, however the device keeps connecting to another repeater which is more distanced with a worse coverage.
The result is that the DTU's WIFI link quality isn't really good (and stable). In average the link quality is -80db / below 50%. 

I have another esp in that cabinet which is running ESPHome and where I configured to use the Accesspoint which is installed in the cabinet. The Wifi link is really good, -15db. 

### Describe the solution you'd like

The UI should support entering a preferred access point by defining the MAC address into a text field. The text field should also support basic checks, e.g. the Mac address should be in the right format. e.g. 0C:72:74:C6:22:6E.

The UI should also support a fallback in case the connection to the AP can not be made any longer. In that case the ESP should connect to any AP in the WIFI/Mesh.

Mockup of the settings: 

[image]

### Describe alternatives you've considered

I brought the AP as close as possible to the ESP, however it keeps connecting to other, more distanced APs.

stefan123t, 2024-08-15:
@swingstate thanks for the nice mockup of your solution to fix the OpenDTU on using only one Access Point matching a specific BSSID / MAC Address in the local 802.11k,r,v Roaming WLAN / Mesh.

This could be used to steer the OpenDTU ESP32 WiFi to connect to the right / best Access Point nearby. 
Note that this effectively would disable WiFi Roaming and (Fast/) Basic Service Set (BSS) transitions at all.
But maybe this is a good fallback / workaround until we have implemented the full 802.11rvk Roaming examples given by Espressif in the ESP-IDF and a working failover strategy

See these two posts for more details for 802.11krv in ESP-IDF:
* https://github.com/tbnobody/OpenDTU/issues/2202#issuecomment-2287191167
* https://github.com/tbnobody/OpenDTU/issues/2202#issuecomment-2289412118

swingstate, 2025-01-28:
Since approx. beginning of January 2025 it appears to me that the WIFI connection of my DTU is less stable, respectively the DTU picks an AP which is far away ( 10m, 3 concrete walls) and the connection drops to 1 Mbit or the DTU disconnects completely for 1 - 3hrs. Close to the DTU, 3m away behind one wall is another AP and if the DTU picks that one the connection is at 65/72 Mbit. All WIFI devices are AVM based. 

Not sure if this is related to any of the updates discussed [here](https://github.com/tbnobody/OpenDTU/issues/2202#issuecomment-2289412118) but it appears to me that the wifi connection of my DTU is less reliable since a couple of weeks and fixing the DTU to a specific BSSID/MAC would probably address the instability in my scenario.