tbnobody/OpenDTU #1359: [Request] device hierarchy im Home Assistant

CFenner, 2023-09-19:
### Is your feature request related to a problem? Please describe.

I have a microinverter with two solar panels and I get a lot of entities on the device view. 

### Describe the solution you'd like

I would like to make use of device hierarchies in home assistant to see dtu, inverter, strings separated like here for NAS and disks in my Synology. 

[image]
[image]

There is the ‘via_device’ field in the [device registry](https://developers.home-assistant.io/docs/device_registry_index/#device-properties) which is used for this relation.

CFenner, 2023-09-20:
First very basic approach..

https://github.com/CFenner/OpenDTU/pull/1/files

Inverter View:
[image]

AC View:
[image]

String View:
[image]

tbnobody, 2023-12-17:
Implemented in 84248ec9b65e18a224a77d02d273bc0fb73548f0

closed 2023-12-17 by tbnobody

CFenner, 2023-12-18:
What about carving out the strings as well?