tbnobody/OpenDTU #1361: Farbe des Info-Balken verändert sich nicht richtig.

Juergen2453, 2023-09-20:
### What happened?

Die Farbe des Info-Balken wird erst ROT wenn die Morgendämmerung erreicht ist, bis dahin bleibt sie Weiß

[image]

[image]

[image]

### To Reproduce Bug

Aktuelle Version

### Expected Behavior

Aktuelle Version

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.9.18

tbnobody, 2023-09-20:
Grau bedeutet, dass keine Daten vom Inverter geholt werden. Dies geschieht z.B. wenn du die Option "Keine Daten Nachts abrufen" aktiviert hat. Wenn dann der Modus von Nacht auf Tag umschaltet und Daten gepollt werden, der Inverter aber noch nicht erreichbar ist wird der Balken rot. 
Also alles wie erwartet...

Juergen2453, 2023-09-20:
Kann es sein das das er früher (ältere Versionen) schon Abends Rot wurde wenn er nach ein paar Minuten nicht mehr erreichbar war?

tbnobody, 2023-09-20:
Die graue Farbe wurde mit acf413b616e737d2e958d8ebcd3a88d79f64d504 am 03. April eingeführt.

Juergen2453, 2023-09-20:
Ich wusste das irgend etwas anders ist. Bin gerade nach Hause gekommen und die Statuszeile ist ROT.
Was habe ich gemacht. Vorher waren 2 Inverter dran und jetzt habe ich nur einen Inverter dran.
[image]

tbnobody, 2023-09-20:
Schaue doch mal unter Info --> NTP
wann beginnt der Nacht Modus? Und hast du jetzt das  fetching bei Nacht  an oder aus? Ohne diese Info ist das  nur rätselraten.

tbnobody, 2023-09-20:
> Bin gerade nach Hause gekommen und die Statuszeile ist ROT.

Das ist korrekt. Zu der Zeit war der Wechselrichter nicht erreichbar und es war Tag Modus und du hattest Daten auch nachts abrufen auf "aus". Daher ist Rot i.O.
Jetzt da Nacht Modus aktiv ist sollte der Balken grau sein.

Juergen2453, 2023-09-20:
Danke erstmal. Ich hoffe ich habe die Logik jetzt verstanden.

tbnobody, 2023-09-20:
Grau = Inverter wird nicht abgefragt
Rot = Inverter wird abgefragt aber antwortet nicht
Gelb = Inverter wird abgefragt, antwortet aber die Leistung ist 0W
Grün = Blau antwortet und produziert Strom

Ob der Inverter abgefragt wird oder nicht kann einfach über die beiden Schalter pro Inverter festgelegt werden.